slipx06/sunsynk-power-flow-card #234: Compact mode: missing extra info with 4 loads

ragg987, 2024-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The "extra" information on loads works well with 2 extra loads, but with 4 extra loads the information is missing. This is in Compact mode. I suspect this is due to the space being reduced, however there seems to be plenty of room in the card to allow this to fit - e.g. by moving the load icons a bit further away from the house.

[image]

### Expected behaviour

display the extra info with 4 loads

### Card Version

4.13.0

### Home Assistant Version

2024.2.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
show_battery: true
show_inverter: false
large_font: true
battery:
  energy: 10200
  shutdown_soc: 1
  show_daily: true
  max_power: 4700
  auto_scale: true
  invert_power: true
  show_absolute: false
  hide_soc: false
  colour: green
solar:
  show_daily: true
  mppts: 1
  pv1_name: ' '
  auto_scale: true
  display_mode: 2
  dynamic_colour: true
  max_power: 6400
load:
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  essential_name: Essential
  additional_loads: 4
  load1_name: ASHP
  load1_icon: mdi:heat-pump
  load4_name: Eddi
  load4_icon: mdi:water-boiler
  load3_name: Car
  load3_icon: mdi:car
  load2_name: Appl's
  load2_icon: mdi:dishwasher
  show_aux: false
  aux_name: aa
  aux_load1_name: aux1
  show_daily_aux: false
  aux_load2_name: aux2
  aux_type: ''
  aux_load1_icon: mdi:dishwasher
  aux_load2_icon: mdi:washing-machine
  show_absolute_aux: false
  dynamic_icon: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  auto_scale: true
  invert_grid: false
  energy_cost_decimals: 3
  load1_name: ne1
  nonessential_name: NE
  load2_name: ne2
  additional_loads: 3
  load3_name: ne3
entities:
  inverter_power_175: sensor.sunsynk_inverter_power
  day_battery_charge_70: sensor.myenergi_libbi_battery_charge_today
  day_battery_discharge_71: sensor.myenergi_libbi_battery_discharge_today
  battery_voltage_183: none
  battery_soc_184: sensor.myenergi_libbi_soc
  battery_power_190: sensor.myenergi_libbi_power_ct_internal_load
  battery_current_191: none
  grid_power_169: sensor.myenergi_weedon_libbi_home_consumption
  day_grid_import_76: sensor.myenergi_libbi_grid_import_today
  day_grid_export_77: sensor.myenergi_libbi_grid_export_today
  grid_ct_power_172: sensor.myenergi_libbi_power_ct_grid
  day_load_energy_84: sensor.home_consumption_energy_today
  essential_power: sensor.home_essential_power
  day_pv_energy_108: sensor.myenergi_libbi_solar_generation_today
  pv1_power_186: sensor.weedon_total_solar_power
  pv2_power_187: none
  pv1_voltage_109: none
  pv1_current_110: none
  pv2_voltage_111: none
  pv2_current_112: none
  essential_load1: sensor.myenergi_libbi_power_ct_monitor
  energy_cost_buy: sensor.electricity_22l4295702_1012455614354_current_rate
  essential_load3: sensor.myenergi_zappi_22675349_power_ct_internal_load
  essential_load4: sensor.myenergi_eddi_23504745_power_ct_internal_load
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  environment_temp: sensor.yutaki_1091_temp_outdoor
  essential_load2: sensor.appliances_total_power
  essential_load2_extra: sensor.myenergi_eddi_23504745_active_heater
  aux_load2: sensor.myenergi_eddi_23504745_power_ct_internal_load
  aux_load1: sensor.myenergi_eddi_23504745_power_ct_internal_load
  aux_load1_extra: sensor.hitachi_t_ext
  load_power_L1: sensor.myenergi_eddi_23504745_power_ct_internal_load
  non_essential_load2: sensor.myenergi_eddi_23504745_power_ct_internal_load
  non_essential_load3: sensor.myenergi_eddi_23504745_power_ct_internal_load
  energy_cost_sell: sensor.electricity_22l4295702_1012455614354_current_rate
panel_mode: true
show_grid: true
inverter:
  modern: true
  autarky: energy
  auto_scale: true
  model: ''
view_layout:
  position: main
dynamic_line_width: true
```

slipx06, 2024-02-12:
Not really a bug. Was a deliberate design decision due to space constraints. Will take another look

ragg987, 2024-02-12:
Ah OK - appreciate your taking another look. Here is my compact layout, it seems there is quite a bit of vertical room on the demand side (above and below the house) that could be used.

I also like the X-shape you use in the other layout types, it moves the focus away from the central inverter and puts it on the 4 legs of info.
[image]

Roving-Ronin, 2024-02-15:
@slipx06 no chance you'll add a 'wide' mode to the card, allowing people to have it covering the width of their screen and then able to also show multiple inverters and batteries? :-)