slipx06/sunsynk-power-flow-card #236: Battery Options - SOC

matomatusov, 2024-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

It wouldn't have to change here if SOC changes.

### Expected behaviour

Question to battery options. Would it be possible to enter an entity here?

[image]

### Mode

Both

### Context / Reason

It must be rewritten if there is a change.

slipx06, 2024-02-13:
Hi Yes you can provide a sensor for this but need to switch to the code editor and add it.

matomatusov, 2024-02-13:
Hi, thank you, I didn't think of that.

closed 2024-02-13 by matomatusov