slipx06/sunsynk-power-flow-card #237: Deye inverters

sarzyk, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Is there any chance for deye hybrid 3ph inverters compatibility?

Good project. Big thanks 👍

### Expected behaviour

3ph deye hybrid inverter compatibility 

### Mode

lite

### Context / Reason

Big thanks

slipx06, 2024-02-14:
HI. The card has support for 3 phase inverters and should work fine with the Deye 3ph as it is very similar to the SunSynk. What is not working for you?

sarzyk, 2024-02-14:
Is working well. Thank for that. But I newbie in Home Assistant and I
correctly understand every write in config I need change from sunsynk to
deye?