slipx06/sunsynk-power-flow-card #250: Please add Brazilian Portuguese language support

slipx06, 2024-02-16:
### Discussed in https://github.com/slipx06/sunsynk-power-flow-card/discussions/249

<div type='discussions-op-text'>

<sup>Originally posted by **R0DRIG0MEL0** February 16, 2024</sup>
{
    "common": {
        "description": "Um Card para o Home Assistant que ajuda a emular o fluxo de energia exibido na tela do inversor",
        "version": "Versão",
        "runtime_to": "TEMPO DE FUNCIONAMENTO",
        "battery_floating": "FLUTUAÇÃO DE BATERIA",
        "charge": "CARGA",
        "to": "PARA",
        "daily_load": "CONSUMO DIÁRIO",
        "daily_aux": "AUXILIAR DIÁRIO",
        "daily_grid_buy": "CONSUMO DIÁRIO DA RERE",
        "daily_grid_sell": "PRODUÇÃO DIÁRIO PARA REDE",
        "daily_solar": "SOLAR DIÁRIO",
        "daily_solar_left": "PRODUÇÃO DIÁRIA / RESTANTE DIÁRIO",
        "total_solar_generation": "PRODUÇÃO DIÁRIA / PREVISÃO SOLAR",
        "daily_charge": "CARGA DIÁRIA",
        "daily_discharge": "DESCARGA DIÁRIA",
        "ratio": "Proporção",
        "timer_off": "Temporizador desligado",
        "timer_on": "Temporizador ligado",
        "priority_batt": "Prioridade da Bateria",
        "priority_load": "Prioridade do Carga",
        "nonessential_name": "Desnecessário",
        "aux_name": "Auxiliar",
        "pv1_name": "PFV1",
        "pv2_name": "PFV2",
        "pv3_name": "PFV3",
        "pv4_name": "PFV4",
        "essential": "Essencial",
        "autarky": "Autoconsumo",
        "discharging": "Descaregando",
        "charging": "Caregando",
        "waiting": "Aguardando",
        "standby": "Em espera",
        "no_battery": "Sem Bateria",
        "selftest": "Autoteste",
        "normal": "Normal",
        "alarm": "Alarme",
        "fault": "Falta",
        "days": "dìas",
        "hrs": "horas",
        "min": "min",
        "flash": "Flash",
        "check": "Check",
        "idle": "Inativo",
        "exporting": "Exportando",
        "importing": "Importando",
        "ongrid": "On-Grid",
        "offgrid": "Off-Grid",
        "normalstop": " Normal Parado",
        "shutdown": "Desligar",
        "offline": "Offline",
        "running": "Rodando",
        "sleepmode": "Dormindo",
        "grid_name": "Rede"

    },
    "errors": {
        "battery": {
            "bat": "Nenhum atributo de bateria definido",
            "energy": "Inclua o atributo de energia da bateria e seu valor em Wh, por exemplo: 5,32 Potência da bateria: 5320",
            "shutdown_soc": "Inclua o atributo shutdown_soc da bateria e seu valor, por exemplo shutdown_soc: 20",
            "full_capacity": "A capacidade total deve estar entre 80 e 100",
            "empty_capacity": "A capacidade vazia deve ser <= 40",
            "show_daily": "Por favor, inclua os atributos day_battery_charge_70 e day_battery_discharge_71 e os IDs das entidade"
        },
        "solar": {
            "sol": "Nenhum atributo solar definido",
            "mppts": "Por favor, inclua o atributo solar mppts e seu valor; 1,2,3 ou 4, por exemplo, mppts: 2",
            "show_daily": "Por favor, inclua o atributo day_pv_energy_108 e o ID da entidade"
        },
        "grid": {
            "show_daily": "Por favor, inclua os atributos day_grid_import_76 e day_grid_export_77 e os IDs das entidade"
        },
        "essential_power": "O atributo Essential_Power está definido como nenhum. Inclua o atributo inverter_power_175 e o ID da entidade para que o cartão possa calcular esse valor.",
        "nonessential_power": "O atributo nonessential_power está configurado como nenhum. Inclua o atributo grid_power_169 e o ID da entidade para que o cartão possa calcular esse valor.",
        "missing_entity": "Por favor, inclua o atributo e o ID da entidade"
    },
    "config": {
      "title": "Título",
      "title_colour": "Cor do Título",
      "title_size": "Tamanho do Título (px)",
      "decimal_places": "Casas decimais",
      "model": "Modelo",
      "autarky": "autoconsumo",
      "cardstyle": "Stylo do Cartão",
      "large_font": "Largura da fonte",
      "modern": "Moderno",
      "show_solar": "Exibir Solar",
      "show_battery": "Exibir Bateria",
      "show_grid": "Exibir Rede",
      "show_aux": "Exibir Aux",
      "dynamic_line_width": "Largura da linha dinâmica",
      "max_line_width": "Largura máxima da linha",
      "solar": "Solar",
      "mppts": "MPPTs",
      "show_daily": "Exibir Diário",
      "show_daily_aux": "Exibir Diário Aux",
      "animation_speed": "Velocidade de Animação",
      "max_power": "Potência Max",
      "pv1_name": "PFV1 Name",
      "pv2_name": "PFV2 Name",
      "pv3_name": "PFV3 Name",
      "pv4_name": "PFV4 Name",
      "display_mode": "Mode de Exibição",
      "battery": "Bateria",
      "load1_name": "Nome Load1",
      "load2_name": "Nome Load2",
      "load3_name": "Nome Load3",
      "load4_name": "Nome Load4",
      "load1_icon": "Icone Load1",
      "load2_icon": "Icone Load2",
      "load3_icon": "Icone Load3",
      "load4_icon": "Icone Load4",
      "aux_load1_name": "Nome Aux_Load1",
      "aux_load2_name": "Nome Aux_Load2",
      "aux_load1_icon": "Nome Aux_Load3",
      "aux_load2_icon": "Nome Aux_Load4",
      "show_daily_buy": "Consumo Diário da Rede",
      "show_daily_sell": "Produção Diária para Rede",
      "no_grid_colour": "Cor No_Grid",
      "grid_off_colour": "Cor Grid_Off",
      "grid_name": "Nome Grid",
      "colour": "Cor",
      "charge_colour": "Cor Charge",
      "aux_colour": "Cor",
      "aux_off_colour": "Cor Off",
      "panel_mode": "Modo do Painel",
      "invert_grid": "Inverter Valores",
      "shutdown_soc": "Desligamento do SOC",
      "energy": "Enêrgia",
      "auto_scale": "Escala automática",
      "three_phase": "Trifásico",
      "dynamic_colour": "Cor Dinâmica",
      "dynamic_icon": "Ícone Dinâmico",
      "invert_power": "Potência do Inversor",
      "show_absolute": "Exibir Absoluto",
      "hide_soc": "Ocultar SOC",
      "invert_aux": "Inverter Aux",
      "show_absolute_aux": "Exibir absolute_aux",
      "show_nonessential": "Exibir Desnecessário",
      "aux_name": "Nome Aux",
      "aux_type": "Icone Aux",
      "aux_loads": "Aux Loads",
      "essential_name": "Nome Essencial",
      "additional_loads": "Adicional Loads",
      "export_colour": "Cor Expor",
      "nonessential_name": "Nome Desnecessário",
      "nonessential_icon": "Icone Non Essential Name",
      "energy_cost_decimals": "Decimais de custo de energia",
      "show_remaining_energy": "Exibir energia restante",
      "cat_title": {
        "title": "Opções de título",
        "general": "Opções gerais",
        "inverter": "Opções do inversor",
        "solar": "Opções Solar",
        "battery": "Opções da Bateria",
        "grid": "Opções da Rede",
        "load": "Opções Load",
        "entities": "Entidades",
        "aux": "Opções do Aux",
        "energy": "Entidade de Enêrgia",
        "bat": "Entidades da Bateria ",
        "sol": "Entidades de Solar",
        "inv": "Entidades do Inversor",
        "ld": "Entidade do Load",
        "gri": "Entidade da Rede"
      }
    }
}</div>

slipx06, 2024-02-17:
@Pho3niX90 can you help with this? The Portuguese (pt) translation works but Brazilian Portuguese (pt-BR) is not recognized

https://github.com/slipx06/sunsynk-power-flow-card/blob/1af3f20ce43b0147f90aa3ffe34ff4accbe512c9/src/localize/localize.ts#L29-L30

slipx06, 2024-02-17:
OK nevermind. I figured it out. Some regex that needed to be considered.

https://github.com/slipx06/sunsynk-power-flow-card/blob/1af3f20ce43b0147f90aa3ffe34ff4accbe512c9/src/localize/localize.ts#L33-L36

closed 2024-02-17 by slipx06

Pho3niX90, 2024-02-17:
Only saw now. Seems you got it working 💪🏻

commit e1954496 (2024-02-16): locale: add Brazilian Portuguese closes #250
commit bf3ad2ee (2024-02-17): fix: remove console logging, fix pt-BR translation fixes #250