slipx06/sunsynk-power-flow-card #265: Display Load and Grid current on card like solar, battery and inverter

codifierr, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently, the card doesn't show load and grid current
[image]

### Expected behaviour

The card should show the load and grid current.

### Mode

Both

### Context / Reason

The card shows Battery, solar, and Inverter. It can also show the Load and Grid current.

slipx06, 2024-02-23:
Yes, the problem is there is not enough free space when some of the other features are enabled i.e. 3 phase, AUX, non-essential, so you have to design conditional checks to only display this data when some of those options are not enabled. Will keep it in mind for a future release

codifierr, 2024-02-24:
Thanks for considering this request.

Roving-Ronin, 2024-03-11:
@codifierr you could always use that box to display the grid Voltage, Frequency and Load (assume you mean the overall total kW being used in the house from all sources?). Just use the applicable sensors for those instead of the default ones.

[image]

This is what I do for the Huawei inverters, in the screenshot above, we can see:

242.6V Grid Voltage
50Hz Frequency
9.48kW Total House Load  (then looking at other boxes, quickly it can be seen that 9.38kW is coming from solar [or battery] and 93W is coming from the grid).