slipx06/sunsynk-power-flow-card #266: it is not dynamic color

matomatusov, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi,
dynamic color is missing in the full version.

[image]

### Expected behaviour

correction of the line thickness

### Card Version

4.16.1

### Home Assistant Version

2024.2.3

slipx06, 2024-02-23:
Do you mean the dynamic Line width is not applied to that line?

matomatusov, 2024-02-23:
yes exactly like that.

closed 2024-02-25 by slipx06

slipx06, 2024-02-26:
Yes the thickness is calculated based on the 2.06kW number i.e. config.entities.grid_power_169

matomatusov, 2024-02-26:
Question: when I entered power L1, L2 and L3 for the three-phase converter, I have a large number for Autarky. what could be the error?

[image]

Thanks

matomatusov, 2024-02-26:
If I don't fill in the power L1, L2 and L3, I still don't know what the 6.82 kW is. The consumption of the house is 3.41 kW.

[image]

matomatusov, 2024-02-26:
entities:
  use_timer_248: input_boolean.sunsynk_use_timer_248
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.solax_inverter_voltage
  inverter_current_164: sensor.solax_inverter_current
  load_frequency_192: sensor.solax_inverter_frequency
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.solax_run_mode
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  grid_power_169: sensor.solax_house_load
  day_grid_import_76: sensor.solax_grid_import
  day_grid_export_77: sensor.solax_grid_export
  grid_ct_power_172: sensor.solax_measured_power_l1
  essential_power: none
  nonessential_power: none
  aux_power_166: sensor.sunsynk_aux_power
  pv1_power_186: sensor.solax_pv_power_1
  pv2_power_187: sensor.solax_pv_power_2
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  pv2_voltage_111: sensor.solax_pv_voltage_2
  pv2_current_112: sensor.solax_pv_current_2
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  environment_temp: sensor.iakovi2_temperature
  remaining_solar: sensor.solcast_pv_forecast_forecast_today
  total_pv_generation: sensor.solax_total_yield
  pv_total: sensor.solax_pv_power_total
  battery_temp_182: sensor.solax_battery_temperature
  inverter_voltage_L2: sensor.solax_inverter_voltage_l2
  inverter_voltage_L3: sensor.solax_inverter_voltage_l3
  inverter_current_L2: sensor.solax_inverter_current_l2
  inverter_current_L3: sensor.solax_inverter_current_l3
  grid_ct_power_L2: sensor.solax_measured_power_l2
  grid_ct_power_L3: sensor.solax_measured_power_l3
  grid_ct_power_total: sensor.solax_measured_power
  grid_voltage: sensor.solax_grid_voltage
  energy_cost_buy: sensor.innogy_current_price_czk_kwh
  energy_cost_sell: sensor.pce_current_price_czk_kwh
  dc_transformer_temp_90: sensor.solax_inverter_temperature
  radiator_temp_91: sensor.solax_inverter_outside_temperature
  essential_load1: sensor.ambienta_power
  essential_load2: sensor.jablotron_futura_coolbreeze_power
  day_load_energy_84: sensor.solax_day_load_energy
  solar_sell_247: input_boolean.sunsynk_solar_sell_247
  load_power_L1: sensor.solax_measured_power_l1
  load_power_L2: sensor.solax_measured_power_l2
  load_power_L3: sensor.solax_measured_power_l3
  inverter_power_175: sensor.solax_inverter_power

slipx06, 2024-02-26:
The full config and if you mark as code it will format correctly

matomatusov, 2024-02-26:
```
type: custom:sunsynk-power-flow-card
cardstyle: full
card_height: 410px
show_solar: true
battery:
  show_daily: true
  energy: 22400
  auto_scale: true
  invert_power: true
  animation_speed: 8
  full_capacity: sensor.solax_battery_charge_upper_soc
  colour:
    - 240
    - 98
    - 146
  hide_soc: false
  show_absolute: false
  empty_capacity: 10
  show_remaining_energy: true
  max_power: 5000
  shutdown_soc: sensor.solax_selfuse_discharge_min_soc
  charge_colour:
    - 238
    - 184
    - 34
  dynamic_colour: true
  linear_gradient: true
solar:
  show_daily: true
  mppts: 2
  pv1_name: West
  pv2_name: East
  auto_scale: true
  dynamic_colour: true
  animation_speed: 8
  max_power: 9770
  display_mode: 3
  colour:
    - 238
    - 184
    - 34
load:
  show_daily: true
  dynamic_icon: true
  auto_scale: true
  load1_icon: mdi:water-boiler
  additional_loads: 2
  load2_icon: mdi:heat-pump
  load1_name: Amb.
  load2_name: Futura
  animation_speed: 8
  max_power: 8000
  dynamic_colour: true
  colour:
    - 193
    - 103
    - 190
  show_aux: false
  essential_name: House
  show_daily_aux: false
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  auto_scale: true
  invert_grid: true
  dynamic_colour: true
  export_colour:
    - 17
    - 165
    - 228
  colour:
    - 227
    - 104
    - 28
  animation_speed: 8
  max_power: 8000
  no_grid_colour:
    - 253
    - 252
    - 252
  energy_cost_decimals: 2
  grid_off_colour:
    - 182
    - 62
    - 80
entities:
  use_timer_248: input_boolean.sunsynk_use_timer_248
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.solax_inverter_voltage
  inverter_current_164: sensor.solax_inverter_current
  load_frequency_192: sensor.solax_inverter_frequency
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.solax_run_mode
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  grid_power_169: sensor.solax_house_load
  day_grid_import_76: sensor.solax_grid_import
  day_grid_export_77: sensor.solax_grid_export
  grid_ct_power_172: sensor.solax_measured_power
  essential_power: none
  nonessential_power: none
  aux_power_166: sensor.sunsynk_aux_power
  pv1_power_186: sensor.solax_pv_power_1
  pv2_power_187: sensor.solax_pv_power_2
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  pv2_voltage_111: sensor.solax_pv_voltage_2
  pv2_current_112: sensor.solax_pv_current_2
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  environment_temp: sensor.iakovi2_temperature
  remaining_solar: sensor.solcast_pv_forecast_forecast_today
  total_pv_generation: sensor.solax_total_yield
  pv_total: sensor.solax_pv_power_total
  battery_temp_182: sensor.solax_battery_temperature
  inverter_voltage_L2: sensor.solax_inverter_voltage_l2
  inverter_voltage_L3: sensor.solax_inverter_voltage_l3
  inverter_current_L2: sensor.solax_inverter_current_l2
  inverter_current_L3: sensor.solax_inverter_current_l3
  grid_ct_power_L2: sensor.solax_measured_power_l2
  grid_ct_power_L3: sensor.solax_measured_power_l3
  grid_ct_power_total: sensor.solax_measured_power
  grid_voltage: sensor.solax_grid_voltage
  energy_cost_buy: sensor.innogy_current_price_czk_kwh
  energy_cost_sell: sensor.pce_current_price_czk_kwh
  dc_transformer_temp_90: sensor.solax_inverter_temperature
  radiator_temp_91: sensor.solax_inverter_outside_temperature
  essential_load1: sensor.ambienta_power
  essential_load2: sensor.jablotron_futura_coolbreeze_power
  day_load_energy_84: sensor.solax_day_load_energy
  solar_sell_247: input_boolean.sunsynk_solar_sell_247
  load_power_L1: sensor.solax_measured_power_l1
  load_power_L2: sensor.solax_measured_power_l2
  load_power_L3: sensor.solax_measured_power_l3
  inverter_power_175: sensor.solax_inverter_power
large_font: true
panel_mode: false
show_battery: true
show_grid: true
inverter:
  model: solax
  three_phase: true
  auto_scale: true
  modern: false
  autarky: power
  colour:
    - 244
    - 241
    - 236
decimal_places: 2
dynamic_line_width: true
max_line_width: 5

```

slipx06, 2024-02-26:
Ok. I found an issue with that formula. I will push a fix.  If you have  `three_phase: true` and you do not provide a sensor for `essential_power` then it should use the formula `essential_power =  load_power_L1 +  load_power_L2 +  load_power_L3` 
There was a problem with the logic so it was using the single phase formula (inverter_power_175 + grid_power_169 - aux_power)

If you have a sensor that measures essential_power then you can explicitly define it in the config i.e. `essential_power: sensor.solax_house_load`

slipx06, 2024-02-26:
I've corrected in 14.6.5. 

`aux_power_166: sensor.sunsynk_aux_power`  <-------- delete this sensor if not used
`full_capacity: sensor.solax_battery_charge_upper_soc` <------------- delete this it is no longer used
`empty_capacity: 10` <-----------delete this it is no longer used

Its not correct to use the same sensors for grid and load

```
grid_ct_power_L2: sensor.solax_measured_power_l2
grid_ct_power_L3: sensor.solax_measured_power_l3
grid_ct_power_total: sensor.solax_measured_power
load_power_L1: sensor.solax_measured_power_l1
load_power_L2: sensor.solax_measured_power_l2
load_power_L3: sensor.solax_measured_power_l3
```

These should be different measurement points.

matomatusov, 2024-02-26:
Then probably like this:

grid_ct_power_L2: sensor.solax_measured_power_l2
grid_ct_power_L3: sensor.solax_measured_power_l3
grid_ct_power_total: sensor.solax_measured_power
load_power_L1: sensor.solax_power_l1
load_power_L2: sensor.solax_power_l2
load_power_L3: sensor.solax_power_l3

slipx06, 2024-02-26:
What integration are you using?

matomatusov, 2024-02-26:
[homeassistant-solax-modbus](https://github.com/wills106/homeassistant-solax-modbus) by Wills 106

slipx06, 2024-02-26:
No they are legacy sensors for the display of the full and empty battery icons

commit 0fe2bca8 (2024-02-25): fix: missing dynamic line width closes #266