slipx06/sunsynk-power-flow-card #267: How to handle two solar inputs?

bensebborn, 2024-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

We have a Sunsynk with battery and solar which shows fine 

however we also have some additional solar (Solis) that isn’t connected to the Sunsynk. As such on other monitoring apps it’s just seen as reducing the house load before the Sunsynk sees it

eg

House load: 2kW
Solis Solar: 1kW
Net house load: 1kW

Sunsynk battery: 1kW
Grid: 0

I’ve setup the card to show the Solar correctly on the Sunsynk but I think if I add the Solis Solar to the sooar entities the calculations are wrong. 

So is it best to add the Solar as a house load instead?

thanks. 

### Expected behaviour

Additional solar can be shown corrextly

### Mode

Both

### Context / Reason

Being able to show solar directly connected to main inverter plus additional solar from existing inverter.

bensebborn, 2024-02-25:
Here’s a comparison between our MyEnergi setup (icon at 5pm is the Sunsynk, icon at 4pm is Solis)

Myenergi is correct

Flow card
[image]
 seems to under report house usage.

bensebborn, 2024-02-25:
[image]

bensebborn, 2024-02-25:
[image]

slipx06, 2024-02-25:
You can try add it as an AUX input and let the card calculate the essential power (i.e don't provide an `essential_power` sensor).  It will use the formula 

essential = inverter_power_175 + grid_power_169 - aux_power_166