slipx06/sunsynk-power-flow-card #282: Huawei - Allow using Battery Sensors for Battery Entities ' & Add max_discharge_power entity to card

Roving-Ronin, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi @slipx06,

Just noticed that there is an entity in the card for the battery called 'shutdown_soc_offgrid:'  But in the documentation it says:

"Only applies for Goodwe inverters. The offgrid battery shutdown percentage used to calculate remaining runtime. Numeric value or sensor i.e. sensor.offgrid_battery_capacity_shutdown" 

### Expected behaviour

Would it be possible to extended this entity to be available to Hauwei users also ?

### Possible Solutions

Would it be possible to extended this entity to be available to Hauwei users also, as looking at the Huawei battery settings and there are two sensors for a Huawei LUNA battery:

1) Is 'number.battery_backup_power_soc' and this could be used to directly feed this information to the card's 'shutdown_soc_offgrid:'  entity, with it enabled for Huawei users also.

Note:  Huawei also has a sensor for 'normal' SOC also that is 'number.battery_end_of_discharge_soc', and can be used to provide the cards 'shutdown_soc:' entity.

Part from updating  'shutdown_soc_offgrid:' entity, would it also be possible to allow Huawei users to enter a sensor for:

1) max_power:  as Huawei has 'the sensor number.battery_maximum_discharging_power'' that would allow pulling this figure directly from the battery into the card.

Lastly I don't think the card has an entity for it already, but if there was one added for:
2) max_discharge_power:   Hauwei / WLCRS can also publish this as a sensor called 'number.battery_maximum_charging_power'

### Mode

Both

### Context / Reason

If this would be possible it would be appreciated and would allow these sections of the card to be 'dynamically' updated from the Huawei BMS / HA side of things, means that even if these settings change, the card would always be correctly configured and charge/discharge etc times would be correct (now if only there was a 'here's the current full capacity of this battery in watts' sensor in the Huawei modbus register I could grab.   ;-)

Thanks. Justin

slipx06, 2024-03-09:
Hi @Roving-Ronin 

I can make the `shutdown_soc_offgrid:` feature available for Huawei users.  

For the other two sensors below you can add these to one of your forks as it will require an update to the logic and data to test

`max_power:`
`max_discharge_power:`

Roving-Ronin, 2024-03-10:
Hi @slipx06 OK let me know re the max_power and max_discharge_power, assuming your saying there'll be a dev release of the card I'll need to test with ?

slipx06, 2024-03-10:
Added `shutdown_soc_offgrid:` to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.21.0

For the other two please create a pull request with the suggested changes.

Roving-Ronin, 2024-03-11:
@slipx06 thanks for that. Re:

max_power:   just trying to understand the card's code and wondering does it actually need any change to support using 

```
max_power: number.battery_maximum_discharging_power
```
versus the current method of:
```
max_power: 5000
```

Looking at the other battery sensors that support calling a sensor reading, they don't appear different from the max_power?

Re the 'max_discharge_power:' and adding that functionality...  way beyond my skills (or at least without pointing in the direction of the file to edit and examples of other entities I can work out from them what/how to change).

My thinking re these though, was that IF an inverter had modbus registers (and from that HA sensors) for max_power / max_discharge_power / energy etc, that would make it easier to configure for most users. Instead of working out the values just reference the sensors and 'auto configured' no matter if they bought the 5, 10, 15, 20kWh etc model. If battery providing this information it would also reflect the actual capacity of the battery, after its used for a few years and can hold less(er) charge.

slipx06, 2024-03-11:
OK the battery `max_power:` is a quick fix. I'll add that for you