slipx06/sunsynk-power-flow-card #285: I have 2 Inverters and 2 Batterys i need a second Battery

elilei123, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hello, i have 2 Inverter sytsems and 2 Battery Sytems i need a Second Battery function 

### Expected behaviour

i need a Second Battery function 

### Mode

full

### Context / Reason

i need a Second Battery function

Roving-Ronin, 2024-03-06:
@elilei123 pending if a 'wide' version of the card is developed, that would have enough space to show multiple inverters / batteries, the option you have it to write a bunch of template sensors that add the battery input/output/daily kWh in and out and others that give the temperature and status.

Suggest you look at these just done for Huawei inverters and batteries, where there's some logic also to say 'show the higher of the two battery temperatures' or 'combine the status but if one has an issue, show the status of the unit with the issue as the output on the card'.

https://github.com/slipx06/sunsynk-power-flow-card/blob/master/docs/examples/huawei_packages/huawei_derived_sensors.yaml

slipx06, 2024-03-17:
Hi. Its not possible at the moment due to limited space.  I might look at creating a wide version in the future. For now you need to use template sensors.

closed 2024-03-17 by slipx06