slipx06/sunsynk-power-flow-card #299: Hooks, Templatable Values, Spacing and Decimals

devilliersjohnny, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Not to sure if the first point goes under **_feature request_** or **_bug_**, but the rest are **_feature requests_** so all are going to be listed here

1. Prepaid units 
    * There are instances where the info cannot be seen because of spacing issues
    * Some utility meters provide two decimal spaces
 
[image]

2. Value other than null
    * Home Assistant doesn't like it if I template a device to equal "null" or null when it doesn't have a value or the sensor is not available/set is there a chance that you could create a different value within your card to equal null (i.e "unavailable") so that that the string value would be replaced with null and there by you would not have to create toggles options to show/hide the entities as they are automatically hidden when they have a null value
  
[image]

[image]

3. Hooks or a way to manipulate second level items that aren't entities 
    * I am not to sure on how to manipulate the second level options that are not entities have attempted to place a input_number as the option for the solar's mppts and it then defaults to 4 mppts on the card
[image]

 How would one go about inputting a value from back end into these?

### Expected behaviour

[image]

### Mode

lite

### Context / Reason

My reasoning behind this would be that you would be able to set up a template and therewith modify the cards front end with the like of `stack-in-card` 'card-mod' 'template-card' which do not have front end editors allowing you to create beautiful dashboards, this would mean that where you would have to dig through the .yaml front code to modify anything you could simply locate the correct template and modify its sensor?

Pho3niX90, 2024-03-10:
For #2, you are setting the value to a string 'null', you can either try the primitive value of null, or better would be undefined.

For #3, these values typically doesn't change often, so a sensor isn't required or typically provided by most inverters.HA doesn't have native templating within a card, but there are templating plugins you can install to use. But I suppose this one could follow the same logic as the batteries power (accept value ot sensor)

slipx06, 2024-03-10:
HI It looks like you are rendering your card in Firefox. For some reason it does not respect the layout options. Try using chrome or edge and the card will render as intended. 

For point 1 I only display 1 decimal for prepaid units. (`${this.toNum(state_prepaid_units.state, 1)}`) I don't really see the benefit of showing more than that but at the end of the day its a user preference.

For point 2 the card makes use of a toNum function to handle sensor data. If this is not a number (i.e. a string as you have it defined) it will return 0. 

```
toNum(val: string | number, decimals: number = -1, invert: boolean = false): number {
        let numberValue = Number(val);
        if (Number.isNaN(numberValue)) {
            return 0;
        }
        if (decimals >= 0) {
            numberValue = parseFloat(numberValue.toFixed(decimals));
        }
        if (invert) {
            numberValue *= -1;
        }
        return numberValue;
    }
```

For point 3. Some of these option accept sensor data (where it makes sense). Most of them are set through the values provided in the config. The documentation is long but it is mentioned there. If there are other options that would better be set through a sensor let me know and I'll take a look at adding it.

devilliersjohnny, 2024-03-10:
@slipx06 -  Thanks for getting back to me

1. The decimals are a nice to have... I have created a entity on the settings ( see below ) that gives this data and allows for the manipulation thereof... but am looking to include this in my overview page of my EDB Dashboard so that a user could see this at a glance?!

[image]

2. If I understand the function correctly it's saying that a string will be tested to see if its a number, if it is it will be cast as a float? Or are you saying that I should ensure that it is a number by pre-casting it? If so how do I do that?

3. Please refer to the post just above with the code... HA does not allow UI config in this case... So I was hoping that I could plug into the card at certain points with my templates?

Thank you both for your responses @slipx06  @Pho3niX90

Pho3niX90, 2024-03-11:
@slipx06 I am wondering if we shouldn't use the state object to determine if an entity should be used, rather than a set value?
Would need to identify which sensors would be applicable for this, and perhaps, even use a conditional such as ((state == None || available == None) && timeSince > 1 minute).

@devilliersjohnny Come to think of it, can you set your templates states to None?

Roving-Ronin, 2024-03-11:
Re the battery temperature and not having a numerical value to the sensor, can't you just use 'float(0)'  so it defaults to 0 if no reading in available?

I do this for the derived sensors for Huawei inverters configurations, that also works for multiple inverters/batteries (setup for 2 in example configs). Here's a card with the example config applied and NO battery attached. Battery sensors show zero and no issues/errors:

[image]

devilliersjohnny, 2024-03-11:
@slipx06 like your thinking!

Tested it out...  

[image]

It however does not yield the result of none in the "Developer Tools >> STATES" Tab, tested it with a few different strings and setting it to none seems to be the only one that returns a different value 'unknown'

[image]

Could you do anything with that?

As far as I can determine it is probably because of the way the template is set up... ie. the template includes "unit_of_measurement" and "device_class" probably forcing it into the field of being a number, but 'None' seems to be a special type within the code?

[image]

devilliersjohnny, 2024-03-11:
@Roving-Ronin welcome to the conversation... the problem with that idea is that temperatures are capable of being 0 and therefore the idea of setting 0="none" or "null" would throw out the entire cards logic when it does?!

devilliersjohnny, 2024-03-11:
Apologies my head is all over the place today! 

@Roving-Ronin the idea behind what I wanted to achieve is that the sensor is hidden when it does not have a value so setting the value to 0 would result in there being a reading when there should be?! It is a valid hypothesis but the idea  @Pho3niX90 came up with would allow for the hiding of the entity when misconfigured? That is if he could use what I told him above!

Pho3niX90, 2024-03-11:
> Apologies my head is all over the place today!
>
> @Roving-Ronin the idea behind what I wanted to achieve is that the sensor is hidden when it does not have a value so setting the value to 0 would [...]

That should be fine. the unknown would actually work perfectly. I believe you can set it to a primitive unknown as well. 

Which items would you like to be able to hide? Only battery or others?

devilliersjohnny, 2024-03-11:
@Pho3niX90 addressing the edit first, the option that I cam looking to hide is the temperature not the entire battery... and there is no option to switch for that! But along those lines it would probably force you to create a whole lot of additional switches/entities to handle all the new options and the card setup page would become over crowded.

Setting 'Unknown' to the primitive... as I mentioned above 'None' is the only one that sets the state to 'Unknown' where as 'Unknown'  sets to to 'Unavailable' which is not unique as almost all other strings do the same

Would it be possible to code for all entities that read 'Unknown' to be hidden? ( I know its a big ask )

Pho3niX90, 2024-03-12:
so we could in theory add a hidden config item, named `dynamicConfig` for example, to enable hiding items when the value is invalid (unknown). There are use cases when a sensor could realistically return a state of None or unknown, (during startup, battery shutdown/sleep), so not sure if this should be default behavior.

@slipx06 what do you think?

slipx06, 2024-03-12:
@Pho3niX90 makes sense but this should not be default behavior. How many other examples besides battery temp are there? Seems like quite a bit of effort with little benefit and would require updating all the CSS display properties?

devilliersjohnny, 2024-03-14:
Hi folks apologies have been AFK for the past few days...  So @slipx06 to answer the question above, all of the temperature sensors follow the same pattern ( Battery, Transformer, Radiator and Environmental ). Think about it, the daily grid buy and sell should also follow this pattern??? But then this would open a larger can of worms as to most if not all of your energy totals ( Daily Solar, load, battery charges ans grid buy/sell ) so it might reach further than just the battery temp?

slipx06, 2024-03-14:
I often want to see power and energy values that are `0`. 
I will apply this to the temp sensors as a start and we can then proceed on a case by case basis.

devilliersjohnny, 2024-03-14:
Its not the ability to set them to 0 that should disable them... its the states of thereof that should point to the disabling of the device! Remember that temperatures can be 0 and even go into the negatives so the numerical value should not be the deciding factor

slipx06, 2024-03-16:
@Pho3niX90 we could update the convertToCustomEntity function

`isValid: () => entity?.state !== null && entity.state !== undefined && entity.state !== 'unknown' || false,`

and then for each of the entities we want to check

```
//Check for unknown states
let isValidStateBatteryTemp = stateBatteryTemp.isValid
```

and finally update the CSS display property

`display="${!config.show_battery || !data.isValidStateBatteryTemp ? 'none' : ''}"`

Pho3niX90, 2024-03-16:
> @Pho3niX90 we could update the convertToCustomEntity function
>
> `isValid: () => entity?.state !== null && entity.state !== undefined && entity.state !== 'unknown' || false,`
>
> and then for each of the [...]

That sounds perfect. 

Will quickly have a look today. I want to introduce some unit tests as well so we can easily test new changes.

Pho3niX90, 2024-03-16:
So changes have been done to the compact card, if you could test it out. https://github.com/slipx06/sunsynk-power-flow-card/pull/309/files#diff-6332f9abe6bcdb8a677cd02890c286fe03ad412b0109c49a6f5fcfc5c3d00c45

closed 2024-03-17 by slipx06

slipx06, 2024-03-17:
@devilliersjohnny please test https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.24.1

devilliersjohnny, 2024-03-18:
@slipx06 @Pho3niX90 Thank you! Is working beautifully! Even those that aren't temperature based are disappearing with the 'None' option! What would you guys think about the card auto configuring it self? I mean that if there are entities present it will auto fill the options, for example lets say that you only have two mppts configured, the card then automatically sets the mppt count to two?

Pho3niX90, 2024-03-18:
> @slipx06 @Pho3niX90 Thank you! Is working beautifully! Even those that aren't temperature based are disappearing with the 'None' option! What would you guys think about the card auto configuring it [...]

I syppose we have enough sensors per brand to attempt something like this, wince they are split into their own classes as well now. 

At first setup, and upon choosing the desired brand, we can do a quick check on 3 sensors if it returns data. If so, use that template.

devilliersjohnny, 2024-03-19:
@Pho3niX90 I'm not quite following?! Apologies but I haven't looked art the source code yet, therefore I cannot say that I am familiar with the workings of the card...

PR #309 (2024-03-17): Improve custom entity handling and add more unit handling
This commit improves the handling of custom entities, particularly focusing on ensuring the state is not null, undefined, or unknown. It also adds various system features such as conversion from a state to power and auto conversion from a state to kilowatts. In addition, units of electrical current, energy, and power were added to provide spatial and time recognizability context to the measures. This enhancement results in a more detailed and accurate representation of data. It also includes adjustments in energy conversion for better p [... cut]