slipx06/sunsynk-power-flow-card #306: adding new features

matomatusov, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi,
would it be possible to add the following sensors?
1 -  add solar radiation
2 - add the status of the inverter to the card lite and compoct
3 - add cooling symbol (similar to Timer On)

[image]

### Expected behaviour

adding new features

### Mode

Both

### Context / Reason

adding new features

slipx06, 2024-03-17:
Thanks for the suggestions. 

1. Can you share some sample data (sensor and value)
2. Not planned for now. Will keep in mind.
3. Not planned for now.Will keep in mind.

slipx06, 2024-04-06:
We need space for number in the 1000's. It might work on the lite card but not possible on the full card. Going to move this to the ideas discussion.