slipx06/sunsynk-power-flow-card #319: grid icon overlay

Hitman79, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

As soon as the net switches from import to export, the symbols overlap. What could be the reason for this, do I have the wrong setting?
[image]
[image]

### Expected behaviour

the icons must be hidden while changing

### Card Version

 v4.24.1

### Home Assistant Version

2024.3.1

slipx06, 2024-03-20:
What sensor are you using for `grid_connected_status_194:`

slipx06, 2024-03-20:
I've added better handling of this in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.24.2.
Make sure you sensor provides on of the expected values otherwise if will default to the connected state

```
export const validGridDisconnected = ['off', '0', 'off-grid', 'off grid', 'offgrid']
export const validGridConnected = ['on', '1', 'on-grid', 'on grid', 'ongrid']
```

closed 2024-03-20 by slipx06

EddieSpaghettie, 2024-03-20:
When my Grid is off it doesnt have a line through it. It used to have it but stopped a while back. Maybe its checking incorrect parameter on Lux inverter

EddieSpaghettie, 2024-03-20:
Just upgraded to the new version. Luxpower inverter. As you can see grid is at 0V etc. I turned the main switch off and it doesnt make a line through the grid : Only getting the Alarm Notification. IS there anything else that needs to be set perhaps?
[image]

EddieSpaghettie, 2024-03-20:
ok got it right. made a binary sensor to check status

Hitman79, 2024-03-22:
> What sensor are you using for `grid_connected_status_194:`

I currently have no sensor set because I read out my inverter via SolarAssistant (MQTT), which unfortunately does not have a status sensor that I need here. I have to build a helper here. Thanks for your support

Hitman79, 2024-03-22:
I no longer have any problems with the latest version v4.24.4. Thank you very much