slipx06/sunsynk-power-flow-card #320: Strange behavior when power is low on Non essential load2

adnovea, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

With low power level the display is no correct on Non-essential load 2.
[image]

### Steps To Reproduce

I have notice this when the power level is low.

May this is a side effect of another parameters setting.
I tried to change some but without success. I have no idea which parameters could be concerned.

### Expected behaviour

Should reflect the value of the corresponding entity.

### Card Version

4.24.2

### Home Assistant Version

2024.3.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
lang: fr
panel_mode: false
card_width: 50%
card_height: 505px
large_font: false
title: Système photovoltaïque
title_colour: '#CCCCCC'
title_size: 19px
show_solar: true
show_battery: true
decimal_places: 2
dynamic_line_width: true
max_line_width: 10
inverter:
  modern: true
  colour: lightgrey
  autarky: power
  model: sunsynk
  auto_scale: false
battery:
  energy: sensor.deye_battery_capacity_watts
  shutdown_soc: sensor.deye_battery_shutdown
  invert_power: false
  colour: '#f06292'
  charge_colour: '#88AA88'
  dynamic_colour: true
  linear_gradient: false
  show_daily: true
  animation_speed: 6
  max_power: 2000
  show_absolute: true
  auto_scale: true
  hide_soc: false
  show_remaining_energy: true
solar:
  colour: orange
  show_daily: true
  mppts: 2
  animation_speed: 9
  max_power: 4920
  pv1_name: Ouest
  pv2_name: Est
  auto_scale: false
  display_mode: 2
  dynamic_colour: true
load:
  colour: '#5fb6ad'
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  animation_speed: 8
  max_power: 5000
  essential_name: Secouru
  aux_name: Surplus (V.E.)
  show_absolute_aux: true
  aux_type: default
  aux_colour: lightblue
  aux_off_colour: grey
  aux_load1_name: Véhicule Élect.
  aux_load1_icon: mdi:car-electric-outline
  aux_load2_icon: mdi:garage
  additional_loads: 2
  load1_name: E.C.S.
  load1_icon: mdi:faucet
  load2_name: Autres
  load2_icon: mdi:power-plug-battery-outline
  auto_scale: false
  aux_loads: 2
  aux_load2_name: Autres
grid:
  colour: '#488fc2'
  grid_name: Réseau EDF
  export_colour: yellow
  no_grid_colour: lightblue
  grid_off_colour: rgb(192,0,0)
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  nonessential_icon: default
  nonessential_name: Gros consommateurs
  additional_loads: 2
  load1_name: P.A.C.
  load1_icon: mdi:heat-pump-outline
  load2_name: Autres
  load2_icon: mdi:washing-machine
  invert_grid: false
  animation_speed: 8
  max_power: 6000
  auto_scale: false
  energy_cost_decimals: 0
entities:
  use_timer_248: switch.deye_time_of_use
  priority_load_243: switch.deye_priority_mode
  day_battery_charge_70: sensor.deye_daily_battery_charge
  day_battery_discharge_71: sensor.deye_daily_battery_discharge
  day_load_energy_84: sensor.deye_daily_load_consumption
  day_grid_import_76: sensor.deye_daily_energy_bought
  day_grid_export_77: sensor.deye_daily_energy_sold
  day_pv_energy_108: sensor.deye_daily_production
  day_aux_energy: sensor.deye_daily_aux
  inverter_voltage_154: sensor.deye_grid_voltage_l1
  load_frequency_192: sensor.deye_load_frequency
  inverter_current_164: sensor.deye_current_l1
  inverter_power_175: sensor.deye_inverter_l1_power
  grid_power_169: sensor.deye_internal_ct_l1_power
  grid_ct_power_172: sensor.deye_external_ct_l1_power
  grid_connected_status_194: binary_sensor.deye_grid_presence
  pv1_power_186: sensor.deye_pv1_power
  pv2_power_187: sensor.deye_pv2_power
  pv3_power_188: none
  pv4_power_189: none
  pv_total: null
  battery_voltage_183: sensor.deye_battery_voltage
  battery_soc_184: sensor.deye_battery_soc
  battery_power_190: sensor.deye_battery_power
  battery_current_191: sensor.deye_battery_current
  battery_rated_capacity: 74
  essential_power: sensor.deye_load_l1_backup_power
  essential_load1: sensor.ecs_backup_power
  essential_load2: sensor.deye_nonessential_others
  nonessential_power: sensor.deye_load_l1_home_power
  non_essential_load1: sensor.aquarea_energy_consumption
  non_essential_load2: sensor.deye_essential_others
  pv1_voltage_109: sensor.deye_pv1_voltage
  pv1_current_110: sensor.deye_pv1_current
  pv2_voltage_111: sensor.deye_pv2_voltage
  pv2_current_112: sensor.deye_pv2_current
  inverter_status_59: sensor.deye_running_status
  aux_power_166: sensor.deye_gen_power
  aux_load1: none
  aux_load2: 0
  aux_connected_status: sensor.deye_smartload_enable_status
  remaining_solar: sensor.energy_production_today_remaining
  radiator_temp_91: sensor.deye_ac_temperature
  dc_transformer_temp_90: sensor.deye_dc_temperature
  battery_temp_182: sensor.deye_battery_temperature
  prog1_time: sensor.deye_mode_point_1
  prog1_capacity: sensor.deye_mode_soc_point_1
  prog1_charge: sensor.deye_mode_grid_point_1
  prog2_time: sensor.deye_mode_point_2
  prog2_capacity: sensor.deye_mode_soc_point_2
  prog2_charge: sensor.deye_mode_grid_point_2
  prog3_time: sensor.deye_mode_point_3
  prog3_capacity: sensor.deye_mode_soc_point_3
  prog3_charge: sensor.deye_mode_grid_point_3
  prog4_time: sensor.deye_mode_point_4
  prog4_capacity: sensor.deye_mode_soc_point_4
  prog4_charge: sensor.deye_mode_grid_point_4
  prog5_time: sensor.deye_mode_point_5
  prog5_capacity: sensor.deye_mode_soc_point_5
  prog5_charge: sensor.deye_mode_grid_point_5
  prog6_time: sensor.deye_mode_point_6
  prog6_capacity: sensor.deye_mode_soc_point_6
  prog6_charge: select.deye_mode_grid_point_6
  energy_cost_buy: sensor.linky_papp
  energy_cost_sell: none
  solar_sell_247: sensor.deye_energy_selling
```

slipx06, 2024-03-21:
I'll take a look

adnovea, 2024-03-21:
THX Slipx06 for your commitment into this project. 
That was a quick fix.