slipx06/sunsynk-power-flow-card #322: Solis battery direction broken

Pho3niX90, 2024-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently in the latest version the battery direction seems broken. It uses both the battery power and direction sensor since the power alone is unsigned. 

I am out most of the day, so this is just a reminder. 

### Steps To Reproduce

N/A

### Expected behaviour

Should use both sensors for direction. 

### Card Version

4.24.4

### Home Assistant Version

2024.3.2

### Configuration

```shell
N/A
```

### Relevant log output

```shell
N/A
```

slipx06, 2024-03-23:
Suggest the following changes but you will need to test. 

https://github.com/slipx06/sunsynk-power-flow-card/blob/cccbc7dc21e945a5a2d26d885e8d89d72fe724fd/src/index.ts#L224-L225

to
```
let gridVoltage = !stateGridVoltage.isNaN() ? stateGridVoltage.toNum(0) : null;
let batteryCurrentDirection = !stateBatteryCurrentDirection.isNaN() ? stateBatteryCurrentDirection.toNum(0) : null;
```

and

https://github.com/slipx06/sunsynk-power-flow-card/blob/cccbc7dc21e945a5a2d26d885e8d89d72fe724fd/src/index.ts#L538-L541

to

```
        if (batteryCurrentDirection != null) {
            if (inverterModel == InverterModel.Solis && batteryCurrentDirection === 0) {
                batteryPower = -batteryPower;
            }
        }
```

slipx06, 2024-03-23:
Please test https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.24.5

Pho3niX90, 2024-03-23:
so the issue is actually related to the isNaN function in the custom card, and the logical or (||), when it's false, it uses the right hand operator which is true. So changing this to the null coalescing fixes it, but introduces an issue regarding the toString on the inverterStatus

PR #324 (2024-03-24): Refine custom entity validations and improve unit conversions
This commit refines the validation of custom entity's state - ensuring it is not null, empty, undefined, or 'unknown'. It also revamps unit conversion functions for better accuracy and precision, specifically the conversion to power and optimizing conversions for kilowatt units in InverterModel.Solis. These changes enhance the representational accuracy and clarity of data.

fixes: #322
closed 2024-03-24 by slipx06