slipx06/sunsynk-power-flow-card #325: Add support to MPP Solar

74mzsolt, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

some functionality of the MPP Solar is not working

### Expected behaviour

MPP Solar functions are more supported

### Mode

lite

### Context / Reason

...

Roving-Ronin, 2024-04-02:
@74mzsolt Giving a bit more information would assist the developer. I assume your referring to: https://www.mppsolar.com/v3/

If so you also need to provide him with a link/photo of these inverters (or a model that identifies the brand for the majority of users, not just your own model), as well as also providing him with all the battery and inverter status / alert messages that you inverter may output.

e.g: https://github.com/slipx06/sunsynk-power-flow-card/blob/master/src/inverters/brands/huawei.ts

However this will also depend upon the integration that your using to make the inverter available/visible within HA to start with.   Do you have this already setup and able to read the inverter+battery data?

Side Note: Looking at this inverter and wondering is it just another PIP inverter ?

74mzsolt, 2024-04-04:
hi, thanks for your reply.

This mainly good for many of us, MPP Solar users - (https://www.thunor.eu/en/products/mpp-solar-pip5048mgx) 
Yes, I have a setup and I can read battery data as well from Pylontech Batteries

Roving-Ronin, 2024-04-04:
@74mzsolt as mentioned before, to ask slipx06 to add an inverter to the card, so at a minimum a image of the inverter appears, required you provide a high quality image of the inverter. 

YOU need to do that, I've tried looking myself and this inverter maker doesn't appear to have any, with all their images that are higher quality all having the inverters appearing at an angle (instead of front on).

If you the inverter doesn't work with the card and the expected sensors readings (i.e. card expects say Status = Normal, but your sensors reply with 'OK', then you also need to provide all the status / alarms / conditions etc that your inverter reports to allow them to be coded into the card).  Also linked to this is HOW your accessing the data on the inverter and pulling it into Home Assistant ? As mentioned before, what integration are you using to do that ? You need to be using a documented integration (or setup a GitHub Repo) and create a (good) solution for doing this, as there is no use adding an inverter unless there is a common way for everyone to integrate it into their Home Assistant. If there's isn't a common way for everyone to do that for a particular brand of inverter, the card will only be able to provide the inverter image and no one else will have the correct status / alarms / conditions etc that your inverter reports that then works with the card.

Slipx06 makes a great card and is extremely responsive to requests, but you need to provide him with the required information if he is to add any inverter to the card.

slipx06, 2024-04-05:
Thanks @Roving-Ronin.
@74mzsolt I've managed to find an image of your inverter. I'll add it in the next release.

74mzsolt, 2024-04-06:
You are amazing, thanks a lot

slipx06, 2024-04-06:
I'm moving this to a discussion. If you share the inverter status codes I can add those.