slipx06/sunsynk-power-flow-card #326: Pre-Paid Units - Currency Symbol addition

beccsjb, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently the field for the prepaid_units component is unable to show a currency symbol (without manipulation)

### Expected behaviour

It would be nice to set a defined currency label, ie $0.00

### Mode

Both

### Context / Reason

The reasoning behind this would be to know what the number being presented is refering to

slipx06, 2024-03-25:
Hi. The prepaid units are meant to only display the number of units available. These are independent of currency. I could display a unit of measurement if that helps (assuming you are using this field to show balance). There may not be enough space though,

beccsjb, 2024-03-25:
G'day, ah ok! Yeah i had assumed that this position was to show calculated balance of the power bill.

I must say, this is such an awesome card and thank you so much for your hard work!

closed 2024-03-26 by slipx06