slipx06/sunsynk-power-flow-card #338: Energy consumption of 'non essential load'

blackbird0589, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Essential loads can only be displayed with power, but not with energy consumption entities

### Expected behaviour

Power and energy sensor entity to be displayed in essential loads area

### Mode

full

### Context / Reason

Actually I'm using the non essential loads area for visualization of my wallboxes. Unfortunately there's no option to display the sensor for consumption of essential loads - only the power can be displayed. Is it possible to insert energy sensor area for non essential loads as it is in load section?
Thanks a lot for this great project!

Roving-Ronin, 2024-03-27:
@blackbird0589 Could you please paste an example of the card how your seeing it ? And the inverters your using?

Here's an example from a site, using the setup for a Huawei inverter(s), in the purple card area is the 'Essential Load'.

[image]

As shown there, the box showing 132W / 136W is the Active Power for the Essential Load for the house. In the Huawei setup its slightly different as its based upon stripping out the Non-Essential and AUX power values, but for your question its fine.
Above there is where I'm using to display the Total Energy used for the whole house for the day (Non-Essential + Essential + AUX) , showing 37.7kWh... I presume your referring to that part of the card and getting the Essential_Load kWh used for the day into there?

If your trying to do the same thing, but for your Essential Load only, if your inverter brand doesn't provide a CT reading and monitoring for the Essential_Load by itself, then you need to have some other device to provide it (i.e. IoTaWatt or a Shelly EM etc). Once you have Power readings for that circuit(s) you can use the utility_meter in HA to define a template sensor calling it.

I prefer to use the Energy_Meter addon instead of utility_meter, with the former allowing tracking costs at the same time. 

```
# IMPORT UTILITY METERS
# See: https://github.com/zeronounours/HA-custom-component-energy-meter

energy_meter:
  sunsynk_essential_energy_daily:
    unique_id: sunsynk_essential_energy_daily
    name: SunSynk - Essential - Energy Daily
    source: sensor.name-of-the-sensor-giving-your-essential-power-reading
    cycle: daily
    price_entity: &electricity-price sensor.electricity_price
```

Just swap energy_meter for utility_meter to use that, also the price_entity above is used to call another sensor with the cost of power from the grid at that moment, then work out the accruing cost for that sensor above and have a like names sensor for tracking the cost.  Just remove it if not wanted or if using the utlity_meter instead.

Once you've got that sensor going and showing your Essential_Load kWh used for the day, map it to the entity: day_load_energy_84:

slipx06, 2024-03-27:
@blackbird0589 this should already work. What issue are you having?. You can display any data there i.e. temp, energy power etc

```
essential_load1: sensor.daily_load1_energy
essential_load2: sensor.daily_load2_energy
```

blackbird0589, 2024-03-28:
I'm sorry, but I've made an mistake in my description. I'm using 'non essential load' area for displaying wallboxes and my question is if you could add a sensor for 'non essential loads consumption' like it is in the essential load area.
[image]

slipx06, 2024-03-28:
Ok I understand now.  Yes that cant be displayed at the moment, there just isn't any space.