slipx06/sunsynk-power-flow-card #341: Config Issue

RafSchandevyl, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Grid & Non Essential have always same value.  and other value always 0

[image]

### Expected behaviour

non-essential should be 0

### Card Version

latest

### Home Assistant Version

latest

RafSchandevyl, 2024-03-28:
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  show_daily: false
  invert_power: true
  energy: 15000
  auto_scale: false
  dynamic_colour: false
  show_remaining_energy: true
  linear_gradient: true
  hide_soc: false
  show_absolute: true
  shutdown_soc: 0
  colour:
    - 35
    - 185
    - 24
  charge_colour:
    - 9
    - 61
    - 5
solar:
  show_daily: true
  mppts: 2
  pv1_name: Vooraan
  pv2_name: Achteraan
  auto_scale: false
  dynamic_colour: true
  pv3_name: Unmetered
load:
  show_daily: true
  show_aux: true
  aux_name: Vitovalor PT2
  aux_type: mdi:fuel-cell
  invert_aux: true
  show_absolute_aux: true
grid:
  show_daily_buy: false
  show_daily_sell: false
  show_nonessential: true
  auto_scale: true
  invert_grid: false
entities:
  day_battery_discharge_71: sensor.energy_daily_from_battery_total
  battery_voltage_183: sensor.sunsynk_battery_voltage
  battery_soc_184: sensor.lfx_battery_soc
  battery_power_190: sensor.lfx_battery_power
  battery_current_191: sensor.sunsynk_battery_current
  essential_power: sensor.lfx_power_to_house_total
  pv1_power_186: sensor.lfx_em_solar_production_extra_01
  pv2_power_187: sensor.lfx_power_inverter_solar_production_total
  day_pv_energy_108: sensor.energy_daily_to_house_total_split_solar
  day_battery_charge_70: sensor.energy_daily_to_battery_total
  day_load_energy_84: sensor.energy_daily_to_house_total
  inverter_power_175: sensor.lfx_power_from_inverter
  grid_ct_power_172: sensor.p1_grid_power
  aux_power_166: sensor.lfx_power_production_not_metered_total
  essential_load1: sensor.lfx_power_to_house_total
inverter:
  autarky: power
  modern: false
  model: sofar
  auto_scale: true
  colour:
    - 98
    - 118
    - 91
dynamic_line_width: true
show_grid: true
panel_mode: true
large_font: true

slipx06, 2024-03-28:
Hi 
Those three sensors are:

grid_power_169: <You have no sensor defined>
nonessential_power: <You have no sensor defined>
grid_ct_power_172: sensor.p1_grid_power

Since grid_power169 is not defined it will display `0`
If you dont provide a sensor for nonessential_power it is calculated using the formula

`nonessential = grid_ct_power_172 - grid_power_169` which is why it is the same as grid_ct_power_172

slipx06, 2024-04-03:
This looks to be a configuration issue. Migrating this to a Sofar discussion