slipx06/sunsynk-power-flow-card #355: Extra info for aux load

virtualj, 2024-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, I use aux load for wallbox and to have extra info with car battery percentage I need to configure 1 extra aux load. This situation cause a reduntant info about the power (same value) and the icon of the extra load does not reflect the aux disconnected colour. This is the actual view:
[image]

### Expected behaviour

Will be usefull to have the extra info also when there is no extra load or the extra load is only one, like this:
[image]

### Mode

full

### Context / Reason

I think that configuration with only one aux is tipical and the extra info can be usefull to much people.
Thank you!

slipx06, 2024-04-04:
Thanks for the suggestion. Its possible to add this. May need to shrink the AUX icon to make space. Will take a look

slipx06, 2024-04-04:
Added to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.28.0

closed 2024-04-04 by slipx06

Roving-Ronin, 2024-04-05:
@slipx06 is seems that this change and its not showing aux\_load1\_extra unless aux\_loads: = 0, also means that where AUX is used for >0 that only aux\_load2\_extra will be shown. :-(

pre v4.280.0

[image]

\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_\_

post updating to v4.280.0

[image]

slipx06, 2024-04-05:
@Roving-Ronin sorry about that. Pushed a fix in [4.28.1](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.28.1)