slipx06/sunsynk-power-flow-card #356: Too long texts, daily energy not shown correct and some wishes at the end

Breinholm-coder, 2024-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]

1. I'm using the enlarged text option to make the screen more readable on my mobile. But the Danish translations of the texts are far too long (see lower left side). These texts could be shortened to: "Daglig salg af El" and "Daglig kob af El".
2. The daily Energy values shown above these texts are also wrong. They are not reset daily as intended - only the total values are shown. The same entities are daily restarted correctly on my energy page.
3. I only want to see the power flow (except for my battery state). The big box on the right side of the inverter is therefore obsolete - It would be nice if its presence could be configurable or dependent on the count of declared entities for the box.
4. The box above the inverter is larger than needed for my one line with power. Can its size and presence also depend on the count of declared entities? 
5. It's impossible to see statistics for the sum box from my 4 solar inputs - Can this be added?
6. I've used the load AUX function to add my extra micro-inverter (with inverted power flow) - This works fine. But an equivalent Aux function under the Inverter configuration would be more appropriate.
Thanks / Peter

### Expected behaviour

Shorter translations of the original texts.
Daily energy values are reset at 00.00 hours.
Boxes are shown and sized by the count of declared content (entities).
Curves (statistics) available for the sum of solar inputs. 

### Card Version

V4.27.0

### Home Assistant Version

Core 2024.4.0, Supervisor 2024.03.1, Operating System 12.1, HMI  20240403.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: 'yes'
solar:
  mppts: 4
  pv1_name: Syd 1
  pv2_name: Vest
  pv3_name: Syd 2
  pv4_name: Øst
  auto_scale: true
  colour:
    - 255
    - 249
    - 148
  max_power: 22293
  dynamic_colour: true
  show_daily: true
  display_mode: 3
battery:
  energy: 30000
  shutdown_soc: 15
  show_daily: true
  auto_scale: true
  colour:
    - 177
    - 221
    - 139
  invert_power: false
  show_absolute: false
  show_remaining_energy: false
  max_power: 22000
  dynamic_colour: true
  linear_gradient: true
load:
  show_aux: true
  colour:
    - 254
    - 251
    - 65
  dynamic_colour: true
  show_daily: true
  auto_scale: true
  invert_aux: false
  show_daily_aux: true
  dynamic_icon: true
  essential_name: Bakkevænget 75C
  max_power: 22000
  aux_type: mdi:generator-portable
  aux_name: Micro Inverter
  show_absolute_aux: false
  aux_colour:
    - 255
    - 255
    - 255
  aux_off_colour:
    - 223
    - 238
    - 212
  aux_loads: 0
  aux_load1_name: HM-1200
  aux_load1_icon: mdi:air-filter
entities:
  inverter_grid_voltage_154: sensor.solarman_grid_voltage_l1
  inverter_load_freq_192: sensor.solarman_load_frequency
  inverter_out_164: sensor.solarman_current_l1
  inverter_out_175: sensor.solarman_total_power
  inverter_load_grid_169: sensor.solarman_total_grid_power
  battery_voltage_183: sensor.deye_syd_vest_battery_voltage
  battery_soc_184: sensor.deye_syd_vest_battery_soc
  battery_out_190: sensor.solarman_battery_power
  battery_current_191: sensor.deye_syd_vest_battery_current
  grid_external_power_172: sensor.solarman_grid_external_power
  grid_status_194: sensor.solarman_grid_connected_status
  inverter_status_59: sensor.solarman_running_status
  pv1_power_186: sensor.deye_syd_vest_pv1_power
  pv2_power_187: sensor.deye_syd_vest_pv2_power
  pv3_power_188: sensor.deye_syd_ost_pv1_power
  pv4_power_189: sensor.deye_syd_ost_pv2_power
  pv1_v_109: sensor.solarman_pv1_voltage
  pv1_i_110: sensor.solarman_pv1_current
  pv2_v_111: sensor.solarman_pv2_voltage
  pv2_i_112: sensor.solarman_pv2_current
  pv3_v_113: none
  pv3_i_114: none
  pv4_v_115: none
  pv4_i_116: none
  day_pv_energy_108: sensor.deye_syd_vest_daily_production
  total_pv_generation: sensor.produceret_energi_i_dag
  battery_power_190: sensor.pwr_fra_batterier
  battery_temp_182: sensor.deye_syd_vest_battery_temperature
  day_battery_charge_70: sensor.deye_syd_vest_daily_battery_charge
  day_battery_discharge_71: sensor.deye_syd_vest_daily_battery_discharge
  battery_status: sensor.deye_syd_vest_status_connection
  day_grid_import_76: sensor.energy_consumed_luxembourg
  day_grid_export_77: sensor.energy_produced_luxembourg
  grid_ct_power_172: sensor.pwr_ph1_grid_sum
  grid_ct_power_L2: sensor.pwr_ph2_grid_sum
  grid_ct_power_L3: sensor.pwr_ph3_grid_sum
  grid_ct_power_total: sensor.power_grid_sum
  grid_voltage: sensor.voltage_phase_1
  essential_power: sensor.power_forbrug
  energy_cost_buy: sensor.energi_data_service
  energy_cost_sell: sensor.nordpool_kwh_dk1_dkk_3_10_0
  aux_power_166: sensor.power_mi
  inverter_power_175: sensor.power_deyes
grid:
  show_daily_buy: true
  colour:
    - 255
    - 140
    - 130
  show_daily_sell: true
  auto_scale: true
  no_grid_colour:
    - 247
    - 206
    - 70
  export_colour:
    - 0
    - 128
    - 128
  energy_cost_decimals: 2
  show_nonessential: false
  invert_grid: false
  grid_off_colour:
    - 40
    - 95
    - 244
  max_power: 22170
  additional_loads: 0
inverter:
  modern: true
  three_phase: true
  auto_scale: true
  autarky: power
  colour:
    - 192
    - 192
    - 192
  model: deye
show_grid: true
decimal_places: 1
show_battery: true
large_font: true
panel_mode: false
dynamic_line_width: true
max_line_width: 8
```

slipx06, 2024-04-04:
Hi Thanks for the feedback. You must be the first Danish user 😄 

1. I will update the translated text as suggested: "Daglig salg af El" and "Daglig kob af El". I would be grateful if you could check and make further suggestions. https://github.com/slipx06/sunsynk-power-flow-card/blob/master/src/localize/languages/da.json
2. The card will display whatever the sensor value is. There is no internal logic to calculate daily energy. You should create a utility meter that resets daily based on your energy sensors.
3, 4. This level of customization is not possible at the moment. Something to consider for the future. Doe your integration not provide you with this sensor data? 
5. If you are are referring to the `more-info` dialog this is not possible as the PV total value is calculated by the card and does not contain any historical state information. You ca create you own sensor that sums this and provide it as input for the  `pv_total:` entity. It will then contain historical state information and statistics.
6. Something to consider. Currently grouped under the load section as it made sense when the AUX port was configured to drive a load.

slipx06, 2024-04-04:
No. 1 has been fixed in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.28.0

slipx06, 2024-04-05:
Thanks. The easiest would be to refer to the [English](https://github.com/slipx06/sunsynk-power-flow-card/blob/v4.28.1/src/localize/languages/en.json) translation file and check for accuracy.

slipx06, 2024-04-06:
Thanks for updating the translation. I have merged for next release.

closed 2024-04-06 by slipx06