slipx06/sunsynk-power-flow-card #357: aux_off_colour

EddieSpaghettie, 2024-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I have a lux inverter and aux_off_colour set to orange. When i have load it shows correct but when load is 0 it doesnt change. Not sure if im missing something?

[image]

[image]

[image]

### Expected behaviour

If aux is 0 it should be orange

### Card Version

Latest

### Home Assistant Version

Latest

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
large_font: true
title: null
panel_mode: true
show_battery: true
decimal_places: 2
show_grid: true
dynamic_line_width: true
min_line_width: 1
max_line_width: 6
inverter:
  model: lux
  modern: false
  colour: white
  autarky: energy
  auto_scale: true
battery:
  hide_soc: false
  show: true
  energy: 5000
  shutdown_soc: 20
  colour: '#3abd43'
  show_daily: true
  invert_power: true
  animation_speed: 6
  max_power: 5000
  full_capacity: 99
  empty_capacity: 20
  show_absolute: true
  auto_scale: true
  show_remaining_energy: true
  dynamic_colour: true
  linear_gradient: true
solar:
  show_daily: true
  mppts: 2
  animation_speed: 5
  max_power: 3800
  auto_scale: true
  colour: '#e8ca35'
  pv1_name: 'String A:'
  pv2_name: 'String B:'
  display_mode: 2
  dynamic_colour: true
load:
  aux_off_colour: Orange
  show_aux: true
  show_daily: true
  max_power: 5000
  auto_scale: true
  additional_loads: 2
  load1_name: Geyser
  load2_name: Pool
  load1_icon: mdi:heating-coil
  load2_icon: mdi:pool
  aux_loads: 2
  aux_load1_name: Fridge
  aux_load2_name: Freezer
  aux_load1_icon: mdi:fridge
  aux_load2_icon: mdi:fridge-industrial
  animation_speed: 5
  dynamic_icon: true
  show_daily_aux: true
  dynamic_colour: true
  essential_name: Home
  show_absolute_aux: false
grid:
  colour: '#5490c2'
  no_grid_colour: '#5490c2'
  grid_off_colour: '#5490c2'
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: true
  invert_grid: true
  additional_loads: 1
  max_power: 5000
  dynamic_colour: true
  nonessential_name: Outbuilding
  auto_scale: true
  nonessential_icon: ''
  load1_name: Flat
  load2_name: Pool
  load1_icon: mdi:home-group
  load2_icon: mdi:pool
  grid_name: Eskom
  show_absolute: true
  animation_speed: 5
entities:
  grid_connected_status_194: input_boolean.grid_status
  inverter_voltage_154: sensor.lux_grid_voltage_live
  load_frequency_192: sensor.lux_grid_frequency_live
  inverter_current_164: sensor.battery_discharge_amps_ac
  inverter_status_59: sensor.lux_status
  inverter_power_175: sensor.home_consumption_live
  day_battery_charge_70: sensor.lux_battery_charge_daily
  day_battery_discharge_71: sensor.lux_battery_discharge_daily
  battery_voltage_183: sensor.lux_battery_voltage_live
  battery_soc_184: sensor.lux_battery
  battery_power_190: sensor.lux_battery_flow_live
  battery_current_191: sensor.battery_discharge_amps
  grid_power_169: sensor.lux_grid_flow_live
  day_grid_import_76: sensor.lux_power_from_grid_daily
  grid_ct_power_172: sensor.lux_grid_flow_live
  day_load_energy_84: sensor.energy_to_house_today_sb_totalled
  essential_power: sensor.home_consumption_live
  nonessential_power: sensor.sonoff_10021f7b34_power
  aux_power_166: sensor.fridge_freezer_load
  day_pv_energy_108: sensor.lux_solar_output_daily
  pv_total: sensor.lux_solar_output_live
  pv1_power_186: sensor.lux_solar_output_array_1_live
  pv2_power_187: sensor.lux_solar_output_array_2_live
  pv1_voltage_109: sensor.lux_solar_voltage_array_1_live
  pv1_current_110: sensor.pv1_amps
  pv2_voltage_111: sensor.lux_solar_voltage_array_2_live
  pv2_current_112: sensor.pv2_amps
  radiator_temp_91: sensor.lux_radiator_1_temperature_live
  dc_transformer_temp_90: sensor.lux_radiator_2_temperature_live
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  environment_temp: sensor.openweathermap_feels_like_temperature
  aux_load1: sensor.sonoff_100186cf1f_power
  aux_load2: sensor.sonoff_100186da51_power
  aux_load1_extra: null
  aux_load2_extra: null
  essential_load1: sensor.sonoff_100186ded0_power
  essential_load2: sensor.pool_power
  non_essential_load1: sensor.sonoff_10021f7b34_power
  non_essential_load2: sensor.pool_power
  grid_voltage: sensor.lxp_ba12490088_grid_voltage
  energy_cost_buy: sensor.buy_from_grid_daily
  day_aux_energy: sensor.fridgefreezer_day_total
  battery_temp_182: sensor.lux_battery_max_cell_temperature_live
  battery_rated_capacity: '5000'
  prog1_time: text.lxp_ba12490088_ac_charge_timeslot_1
```

virtualj, 2024-04-04:
aux_off_colour is correlated to the entity aux_connected_status, not to the power. Anyway you can do a sensor template that when aux total load is 0 then binary sensor is equal to 0, either is 1. Also, there is a bug fix in the latest version 4.27.0. Did you do? Anyway it seems that aux_off_colour is not applyed when you set aux_loads more than 0. I'm not sure if it is the expected behaviour or is a bug

slipx06, 2024-04-04:
I should also add this only applies to the AUX icon when you do not have any additional aux loads configured. I probably need to make that clearer in the documentation

EddieSpaghettie, 2024-04-04:
Ok so are you saying that I cant get the colour to change when I have the two loads configured and they are both turned off ?
Will it only work if you have one aux load configured?

slipx06, 2024-04-04:
Hi. No it is only applied to the AUX icon when there are no additional aux loads. Also as mentioned above it uses a separate binary sensor and is not based on power usage.

slipx06, 2024-04-04:
I can look into that.

commit c313812b (2024-04-04): feat: add `aux_dynamic_colour:` closes #357
closed 2024-04-04 by slipx06