slipx06/sunsynk-power-flow-card #379: Some entity does not work as "CLICT" to CHART and ADD one entity more_?

tvarohp, 2024-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]
Hello,
I wanna ask why I can not 
1) click on for exemple temepratura or phases in Load to see the chart as is common in other entities_? this does not work: AUX_load1_extra, AUX_load2_extra, grid_Ct_power_l1,L2,L3,  load_power_L1/L2/L3
2) it is possible to add temperature for additional load icon (in picture as "POOL") = there is not EXTRA entity to fill in. Please

### Expected behaviour

Mentioned above

### Card Version

FULL

### Home Assistant Version

2024.4.2 -12.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
panel_mode: true
cardstyle: full
show_solar: true
inverter:
  modern: false
  model: goodwe_gridmode
  auto_scale: false
  three_phase: true
  autarky: energy
  colour:
    - 5
    - 5
    - 5
battery:
  energy: 10650
  shutdown_soc: 10
  invert_power: false
  colour:
    - 0
    - 4
    - 242
  show_daily: true
  animation_speed: 3
  empty_capacity: 10
  auto_scale: true
  full_capacity: 100
  show_absolute: 'on'
  hide_soc: false
  show_remaining_energy: true
  linear_gradient: true
  dynamic_colour: true
  charge_colour:
    - 0
    - 166
    - 0
solar:
  colour: green
  show_daily: true
  mppts: 2
  auto_scale: true
  max_power: 9645
  pv1_name: DOWN
  pv2_name: UP
  dynamic_colour: true
  display_mode: 2
load:
  colour: orange
  show_daily: true
  show_aux: true
  show_daily_aux: false
  aux_name: Plyn
  auto_scale: true
  load1_name: Pool
  invert_aux: false
  show_absolute_aux: true
  load1_icon: mdi:heat-pump-outline
  dynamic_colour: true
  dynamic_icon: true
  aux_loads: 2
  aux_load1_icon: mdi:water-boiler
  aux_load2_name: Bojler El.
  aux_load2_icon: mdi:fire-circle
  aux_load1_name: Plyn
  load2_icon: mdi:extractor-fan
  load2_name: Bojler
  aux_type: mdi:gas-burner
  essential_name: .
  additional_loads: 1
grid:
  colour: red
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  load1_icon: ''
  load2_icon: ''
  auto_scale: true
  dynamic_colour: true
  export_colour:
    - 242
    - 7
    - 7
  invert_grid: false
  nonessential_icon: mdi:meter-gas
  nonessential_name: Plyn
  additional_loads: 0
entities:
  total_pv_generation: sensor.today_s_pv_generation
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_current_164: sensor.on_grid_l1_current
  inverter_current_L2: sensor.on_grid_l2_current
  inverter_current_L3: sensor.on_grid_l3_current
  inverter_power_175: sensor.sensor_vykonfve_load
  grid_connected_status_194: sensor.grid_mode_code
  day_battery_charge_70: sensor.today_battery_charge
  day_battery_discharge_71: sensor.today_battery_discharge
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  battery_temp_182: sensor.battery_temperature
  battery_status: sensor.battery_mode
  radiator_temp_91: sensor.inverter_temperature_radiator
  dc_transformer_temp_90: sensor.inverter_temperature_air
  grid_power_169: sensor.house_consumption
  day_grid_import_76: sensor.template_denni_3_faze_energy
  day_grid_export_77: sensor.template_denni_3_faze_energy_returned
  grid_ct_power_L2: sensor.active_power_l2
  grid_ct_power_L3: sensor.active_power_l3
  essential_power: sensor.load_loadtuv
  nonessential_power: sensor.pomocnik_denni_spotreba_plyn_kwh_minusova
  aux_power_166: sensor.tuv_temp_heat_0_1_power
  day_pv_energy_108: sensor.today_s_pv_generation
  pv_total: sensor.pv_power
  pv1_power_186: sensor.pv1_power
  pv2_power_187: sensor.pv2_power
  pv1_voltage_109: sensor.pv1_voltage
  pv1_current_110: sensor.pv1_current
  pv2_voltage_111: sensor.pv2_voltage
  pv2_current_112: sensor.pv2_current
  environment_temp: sensor.pws_temp
  day_aux_energy: sensor.pomocnik_denni_spotreba_plyn_kwh_plusova
  day_load_energy_84: sensor.denni_spotreba
  essential_load1_extra: sensor.tuv_up_temperature_temperature
  load_power_L1: sensor.load_l1
  load_power_L3: sensor.load_l3
  aux_load2_extra: sensor.tuv_down_temperature_temperature
  aux_load1: sensor.pomocnik_denni_spotreba_plyn_kwh_plusova
  aux_load2: sensor.pomocnik_tuv_temp_heat_0_energy_denni_spotreba
  energy_cost_sell: '1.5'
  energy_cost_buy: '5.5'
  prepaid_units: '0'
  essential_load2_extra: sensor.tuv_down_temperature_temperature
  inverter_status_59: select.inverter_operation_mode
  grid_ct_power_total: sensor.soucet_fazi
  grid_ct_power_172: sensor.active_power_l1
  pv4_voltage_115: Součet energy_production_today_remaining 1 + 2
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  load_power_L2: sensor.load_l2
  aux_load1_extra: sensor.tuv_up_temperature_temperature
show_battery: true
show_grid: true
decimal_places: 1
large_font: false
dynamic_line_width: false
```

slipx06, 2024-04-11:
Hi. 

1. I have not enabled the more-info dialog for those sensors yet. I will them for you
2. I can enable display of `essential_load1_extra` on the full card when `additional_loads: 1` and `show_aux: true`. Its currently only visible when `show_aux: false`

slipx06, 2024-04-11:
I've added to all of these except load_power_L1, load_power_L2 and load_power_L3. They are rendered as a single text object so not possible

closed 2024-04-11 by slipx06

tvarohp, 2024-04-12:
Thank you so much- it is great!!
Do you have any "Give me caffe" ? i will send you there something :))

slipx06, 2024-04-13:
Pleasure . Sure https://www.buymeacoffee.com/slipx  Thanks

slipx06, 2024-04-30:
@tvarohp I dont see any pictures?

tvarohp, 2024-04-30:
Sorry i added PDF to attachment.

commit 29ed1c84 (2024-04-11): feat: add more clicakble entities closes  #379