slipx06/sunsynk-power-flow-card #380: Add 2 extra batteries

github-daniel-stoian, 2024-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently we can add additional_loads up to 4 in Compact mode, which is very beneficial to see more important consumers in the schema.
[image]

### Expected behaviour

I currently have 2 x Ecoflow DeltaPro which are charged via AC-IN from a Huawei 6KW inverter.
Even if they are not directly connected to Inverter DC batt, they are still part of the ecosystem powering the house during the night.

It will be more suitable to place them as additional_batteries with as additional_batteries up to 2 with SOC, Voltage, Power like normal batteries.
Also which is beneficial to free up the additional_loads slots for other consumers.
[image]

### Possible Solutions

In Compact you can specify:
On battery section : additional_batteries 2
On entities section : 
battery_power_extra1
battery_voltage_extra1
battery_soc_extra1

battery_power_extra2
battery_voltage_extra2
battery_soc_extra2

### Mode

lite

### Context / Reason

It will help to integrate external batteries in the energy flow

github-daniel-stoian, 2024-04-14:
I could create extra 2 cards and place one Ecoflow on one card.

In this case:
-I will have 3 energy cards: hard to arrange on a dashboard, extra management of code
-Ecoflow cards won't have solar, because I don't charge them from solar, only from AC provided by Growatt
-Ecoflow cards mainly will show the battery pictures with SOC, power, voltage ... grid, solar will not be used.

That's why I think they could fix better as additional_batteries in the Huawei inverter card.

slipx06, 2024-04-16:
Hi. Thanks for the suggestion. It's on the to-do list together with multi inverter setups

github-daniel-stoian, 2024-04-18:
Thank you very much. Very good work you are doing.
Probably the ultimate/one Energy card needed on HA.

Roving-Ronin, 2024-05-25:
> Hi. Thanks for the suggestion. It's on the to-do list together with multi inverter setups

Hi @slipx06 any update on this ? I know its a bit of a 'how longs a piece of string question', given time in the day and things to do ;-)

slipx06, 2024-06-01:
Thinking about this more, why not just use two cards next two each other (one for each inverter).

slipx06, 2024-06-20:
Carry on here https://github.com/slipx06/sunsynk-power-flow-card/discussions/374

closed 2024-06-20 by slipx06