slipx06/sunsynk-power-flow-card #389: Added SMA Inverter Model

shingsam, 2024-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Please add SMA Inverter Model.

### Expected behaviour

SMA Inverter model have SMA Sunny Island (Yellow), SMA Sunny Boy (Red), SMA Sunny Try Power (Blue), and SMA Sunny Tripower Smart Energy (White).
I have sample img SMA Sunny Island (Yellow) https://raw.githubusercontent.com/shingsam/SMAVenusDriver/master/assets/overview-inverter.svg

### Mode

lite

### Context / Reason

Many people use your components, and they have different inverter brands.

paolotacc, 2024-04-17:
Can you integrate ABB Power one inverter ?

shingsam, 2024-04-17:
> Can you integrate ABB Power one inverter ?

No, i cant.

slipx06, 2024-04-18:
Can you share some of the inverter status messages that the integration provides

shingsam, 2024-04-18:
> Can you share some of the inverter status messages that the integration provides

35: Fault (Alm), 303: Off (Off), 307: Ok (Ok), 455: Warning (Wrn)

slipx06, 2024-04-20:
Hi. Added to https://github.com/slipx06/sunsynk-power-flow-card/releases/edit/v4.32.0

closed 2024-04-20 by slipx06