slipx06/sunsynk-power-flow-card #399: Unknown Grid Value visible in Compact Mode

limatango79, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]

### Expected behaviour

Make the value invisible

### Card Version

4.32.0

### Home Assistant Version

2024.4.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  energy: 0
  shutdown_soc: 10
  show_daily: false
  auto_scale: true
  show_absolute: false
  invert_power: true
  colour:
    - 85
    - 134
    - 180
  charge_colour:
    - 129
    - 134
    - 156
  dynamic_colour: false
  linear_gradient: false
  hide_soc: false
  animation_speed: 1
  max_power: 1
  show_remaining_energy: false
solar:
  show_daily: true
  mppts: 4
  pv1_name: Nord
  pv2_name: Süd
  pv3_name: West
  pv4_name: Ost
  auto_scale: true
  display_mode: 1
  dynamic_colour: true
  animation_speed: 1
  max_power: 1
  colour:
    - 208
    - 167
    - 78
load:
  show_daily: false
  auto_scale: true
  dynamic_icon: false
  additional_loads: 2
  essential_name: HAUS
  load1_name: EV
  load1_icon: mdi:ev-station
  load2_name: WP
  load2_icon: mdi:heat-pump
  colour:
    - 203
    - 93
    - 93
  dynamic_colour: false
  show_aux: false
  animation_speed: 2
  max_power: 1
  invert_load: false
grid:
  show_daily_buy: false
  show_daily_sell: false
  show_nonessential: false
  auto_scale: true
  invert_grid: true
  grid_name: NETZ
  show_absolute: false
  animation_speed: 1
  export_colour:
    - 63
    - 146
    - 88
entities:
  battery_soc_184: sensor.bat_soc
  battery_power_190: sensor.bat_leistung
  day_pv_energy_108: sensor.pv_heute
  pv1_power_186: sensor.pv_nord_leistung
  pv2_power_187: sensor.pv_sud_leistung
  pv3_power_188: sensor.pv_west_leistung
  pv4_power_189: sensor.pv_ost_leistung
  essential_power: sensor.verbrauch_knx_haus_gesamt_leistung
  essential_load1: sensor.openwb_ladeleistung
  essential_load2: sensor.verbrauch_knx_wp_gesamt_leistung
  grid_ct_power_total: sensor.ac_bezug_einspeisung
  load_frequency_192: sensor.ac_netzfrequenz
  inverter_power_175: sensor.ac_wr_wirkleistung_total
  grid_power_169: sensor.ac_bezug_einspeisung
  solar_sell_247: 'on'
  day_load_energy_84: sensor.verbrauch_heute
large_font: true
panel_mode: false
show_battery: true
show_grid: true
dynamic_line_width: false
inverter:
  three_phase: true
  model: sofar
  modern: true
  autarky: 'no'
  auto_scale: true
  colour:
    - 158
    - 139
    - 167
decimal_places: 1
decimal_places_energy: 0
```

slipx06, 2024-04-23:
Hi 
If you are not using three phase then you need to set  `three_phase: false` and change `grid_ct_power_total: sensor.ac_bezug_einspeisung` to `grid_ct_power_172: sensor.ac_bezug_einspeisung`

slipx06, 2024-04-28:
Closing this as it appears to be a configuration issue.

closed 2024-04-28 by slipx06