slipx06/sunsynk-power-flow-card #401: Excessive decimal places on battery SOC

purcell-lab, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

My Battery SOC is a float which is represented with lots of decimal places.

[image]

### Steps To Reproduce

Assigning a sensor with a float value to battery SOC.

[image]

### Expected behaviour

Would like to see this value rounded in line with the other decimal places elsewhere on the card.

The workaround is a template sensor to convert to a int.

### Card Version

4.32.0

### Home Assistant Version

Core 2024.4.3 Supervisor 2024.04.2 Operating System Frontend 20240404.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
panel_mode: false
large_font: true
card_height: 600
inverter:
  modern: false
  three_phase: true
  auto_scale: true
  autarky: 'no'
  model: solaredge
battery:
  energy: 40000
  shutdown_soc: 1
  show_daily: true
  auto_scale: true
  max_power: 15000
  show_absolute: false
  empty_capacity: 0
  full_capacity: 97
  invert_power: true
  hide_soc: true
  show_remaining_energy: true
  dynamic_colour: true
  linear_gradient: true
solar:
  show_daily: true
  display_mode: 2
  mppts: 2
  pv1_name: West (I2)
  pv2_name: East (I1)
  auto_scale: true
  dynamic_colour: true
load:
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
  show_aux: true
  aux_load1_icon: mdi:car-electric
  aux_load2_icon: ''
  additional_loads: 2
  load1_name: HotWater
  load1_icon: mdi:heat-pump-outline
  load2_name: Fridge
  load2_icon: mdi:fridge
  aux_name: V2H
  aux_type: mdi:car-electric-outline
  load3_name: Fans
  load3_icon: mdi:fan
  load4_name: Appliances
  load4_icon: mdi:devices
  invert_aux: true
  show_absolute_aux: true
  show_daily_aux: true
  aux_dynamic_colour: true
  aux_load1_name: V2H
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  auto_scale: true
  load1_name: HVAC
  load1_icon: mdi:hvac
  load2_name: EV
  load2_icon: mdi:car-electric
  additional_loads: 3
  invert_grid: true
  dynamic_colour: true
  nonessential_icon: mdi:ev-station
  load3_name: Pool
  load3_icon: mdi:pool
entities:
  inverter_voltage_154: sensor.solaredge_m1_ac_voltage_an
  load_frequency_192: sensor.solaredge_m1_ac_frequency
  grid_connected_status_194: binary_sensor.grid_status
  day_battery_charge_70: sensor.powerwall_battery_import_daily
  day_battery_discharge_71: sensor.powerwall_battery_export_daily
  battery_voltage_183: sensor.powerwall_battery_average_voltage_now
  battery_soc_184: sensor.energy_site_percentage_charged
  battery_power_190: sensor.apf_battery_entity
  battery_current_191: sensor.powerwall_battery_average_current_now
  grid_ct_power_L1: sensor.solaredge_m1_ac_power_a
  grid_ct_power_L2: sensor.solaredge_m1_ac_power_b
  grid_ct_power_L3: sensor.solaredge_m1_ac_power_c
  day_grid_import_76: sensor.powerwall_site_import_daily
  day_grid_export_77: sensor.powerwall_site_export_daily
  grid_ct_power_172: sensor.solaredge_m1_ac_power_a
  day_load_energy_84: sensor.powerwall_load_import_daily
  day_pv_energy_108: sensor.powerwall_solar_export_daily
  pv1_power_186: sensor.solaredge_i2_dc_power
  pv2_power_187: sensor.solaredge_i1_dc_power
  pv1_voltage_109: sensor.solaredge_i2_dc_voltage
  pv1_current_110: sensor.solaredge_i2_ac_current
  pv2_voltage_111: sensor.solaredge_i1_dc_voltage
  pv2_current_112: sensor.solaredge_i1_ac_current
  energy_cost_buy: sensor.amber_general_price
  energy_cost_sell: sensor.amber_feed_in_price
  environment_temp: sensor.tewantin_temp
  non_essential_load1: sensor.hvac_power
  non_essential_load2: sensor.ev_power
  non_essential_load3: sensor.pool_power
  grid_voltage: sensor.tesla_wall_connector_grid_voltage
  grid_ct_power_total: sensor.apf_grid_entity
  essential_load1: sensor.hws_power
  dc_transformer_temp_90: sensor.solaredge_i1_temp_sink
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  essential_load2: sensor.fridge_power
  pv_total: sensor.apf_generation_entity
  nonessential_power: sensor.apf_house_entity
  aux_power_166: sensor.essential_power
  essential_load3: sensor.fan_power
  essential_power: sensor.essential_power
  inverter_current_164: sensor.solaredge_m1_ac_current_a
  day_aux_energy: sensor.hws_energy_energy_daily
  grid_power_169: sensor.apf_house_entity
  inverter_power_175: sensor.apf_inverter_power_consumption
  inverter_current_L2: sensor.solaredge_m1_ac_current_b
  inverter_current_L3: sensor.solaredge_m1_ac_current_c
  essential_load4: sensor.oven_power_phase_a
  inverter_voltage_L2: sensor.solaredge_m1_ac_voltage_bn
  inverter_voltage_L3: sensor.solaredge_m1_ac_voltage_cn
view_layout:
  position: main
show_battery: true
show_grid: true
decimal_places: 1
dynamic_line_width: true
```

slipx06, 2024-04-23:
Thanks for reporting. I'll take a look.

slipx06, 2024-04-23:
@Pho3niX90 would you mind checking this. I cant see why this would happen since we call Utils.toNum function which is supposed to handle this

purcell-lab, 2024-04-25:
Thanks, all good now.

https://github.com/slipx06/sunsynk-power-flow-card/assets/79175134/7daa2238-4199-4367-8efd-e2ebe97fdb31

PR #405 (2024-04-24): fix rounding on full card
fixes: #401