slipx06/sunsynk-power-flow-card #404: Powerflow Point at Inverter to additional values running wrong direction

Stepsilein, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The red circued animated powerflow point (see enclosed screenshot) ist running wrong direction, should point togrid also.

### Steps To Reproduce

See my written rest please, thanks and screen here.
[image]

### Expected behaviour

See my explanatiln above please... and screen.

### Card Version

Newest from a Minute ago today.

### Home Assistant Version

Newest from today too.

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
large_font: true
title: Qcells Bermoll 9,35 kWh PV + 6 kWh Akku
title_colour: black
title_size: 18px
show_solar: true
show_battery: true
card_height: 400px
card_width: 100%
decimal_places: 2
inverter:
  modern: true
  colour: black
  autarky: 'no'
  auto_scale: true
  three_phase: false
  model: solax
battery:
  energy: 6000
  battery_typ: Lithium
  shutdown_soc: sensor.solax_selfuse_discharge_min_soc
  invert_power: true
  colour: darkgreen
  show_daily: true
  animation_speed: 9
  max_power: 2500
  full_capacity: 100
  empty_capacity: 10
  show_absolute: true
  auto_scale: true
  show_remaining_energy: true
solar:
  colour: darkorange
  show_daily: true
  mppts: 2
  animation_speed: 10
  max_power: 9450
  pv1_name: String 1
  auto_scale: true
  display_mode: 1
  dynamic_colour: true
  pv2_name: String 2
grid:
  colour:
    - 184
    - 0
    - 0
  show_daily_buy: true
  show_daily_sell: true
  no_grid_colour:
    - 0
    - 0
    - 224
  show_nonessential: false
  show_essential: true
  invert_grid: true
  auto_scale: true
  animation_speed: 10
  dynamic_colour: true
  export_colour:
    - 0
    - 161
    - 0
entities:
  use_timer_248: sensor.solax_grid_mode_runtime
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  day_grid_import_76: sensor.solax_today_s_import_energy
  day_grid_export_77: sensor.solax_today_s_export_energy
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  inverter_voltage_154: sensor.solax_grid_voltage
  load_frequency_192: sensor.solax_grid_frequency
  inverter_current_164: sensor.solax_inverter_current
  inverter_power_175: sensor.solax_inverter_power
  pv1_power_186: sensor.solax_pv_power_1
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  battery_temp_182: sensor.solax_battery_temperature
  dc_transformer_temp_90: sensor.solax_inverter_temperature
  pv2_power_187: sensor.solax_pv_power_2
  pv2_voltage_111: sensor.solax_pv_voltage_2
  pv2_current_112: sensor.solax_pv_current_2
  essential_power: sensor.solax_house_load
  inverter_status_59: sensor.solax_lock_state
  grid_ct_power_172: sensor.solax_measured_power
show_grid: true
load:
  show_daily: false
  auto_scale: true
  dynamic_colour: true
  essential_name: Tim Haus
  dynamic_icon: true
  animation_speed: 9
dynamic_line_width: true
max_line_width: 15
```

### Relevant log output

```shell
No log necessary / available.
```

slipx06, 2024-04-24:
The direction of that dot is determined by the power in the green square. I cant see because the red circle is in the way but this should also be -956W in order for the dot to move towards the grid. If this reading is positive the dot will move away from the grid

Stepsilein, 2024-04-24:
> The direction of that dot is determined by the power in the green square. I cant see because the red circle is in the way but this should also be -956W in order for the dot to move towards the grid. If this reading is positive the dot will move away from the grid

No.
The values are always right, but the dot in der circle moves always from grid side to inverter side, despite power from grid.
In short, the circled dot istcalways movkng right tio left side, if power to grid or from grid (Bezug oder Einspeisung).
Must be somthing wrong at the path between Inverter and that sqare of the additional values.
All the rest ist okay and working nice, but that always moving right to left dot, ist anoing.

slipx06, 2024-04-24:
Thanks The problem is those two values are not the same. The  `grid_ct_power_172:` is negative while `grid_power_169: ` is posiitve . So the dots will always move in opposite direction unless you are drawing power from the grid.  What sensor are you using for  `grid_power_169: ` ?

[image]

Stepsilein, 2024-04-24:
> Thanks The problem is those two values are not the same. The `grid_ct_power_172:` is negative while `grid_power_169: ` is posiitve . So the dots will always move in opposite direction unless you are [...]

This 19 sensor used is (like you could see at the already posted yaml at the beginning):
sensor.solax_measured_power

closed 2024-04-24 by slipx06