slipx06/sunsynk-power-flow-card #412: max_sell_power sensor - enable/disable (goodwe, may be other)

mamo-nick, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Power Limit must be enabled to use the max sell power value. If it is not on, the set value is not worked with. Only when it is turned on, then it follows the set value. This is not taken into account in the graph.

### Expected behaviour

If Power limit is on, then the value will be colored, otherwise gray.

### Possible Solutions

If the sensor is not defined by the user, then the value of max_sell_power will be colored.
If defined, then:

- return value 0 (disable) - max_sell_power will be gray
- return value 1 (enable) - max_sell_power will be colored

https://en.goodwe.com/Ftp/EN/Downloads/User%20Manual/GW_SolarGo_User%20Manual-EN.pdf
see page 36

### Mode

Both

### Context / Reason

better information for users

slipx06, 2024-04-29:
Hi. 

The card has support for this. You can use the `solar_sell_247` entity  i.e. `solar_sell_247: switch.sunsynk_toggle_solar_sell`. It will add an icon shown in the screenshot below. I like the idea of colouring the `max_sell_power` grey when disabled. The icon has a line through when off as well

[image]

mamo-nick, 2024-04-30:
Hi, 
so, that sensor exists. That's nice. Just finish the coloring to gray ;)
Thanks

slipx06, 2024-04-30:
Added to [4.35.0 ](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.35.0)

closed 2024-04-30 by slipx06