slipx06/sunsynk-power-flow-card #413: Energy Period Selectoon

daimonkor, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

sdfd

### Expected behaviour

sfew

### Possible Solutions

sdfdsf

### Mode

lite

### Context / Reason

vdsv

slipx06, 2024-04-29:
Hi. Can you give me an example, not sure I understand the feature request.

daimonkor, 2024-04-29:
Hi, I have the current card https://github.com/flixlix/energy-period-selector-plus (as sample). I think for non current date - show only daily data, other - online

daimonkor, 2024-05-02:
> IMHO seems a bit weird, given this is a POWER flow card, to then have e ENERGY from other days upon it.

Power flow contains daily energy data (but it is power flow card). My idea is only extend logic daily data with help data selection period. So offcourse for prev/next days card can show only daily counters, other - miss.