slipx06/sunsynk-power-flow-card #425: Bigger view

mrfrag1, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, this card is super.  But is there a way the view two set bigger ?
I find the option large font, but there is too small.

Bigger graphics and bigger fonts are perfectly.

Thanks
Martin

[image]

### Expected behaviour

.

### Mode

lite

### Context / Reason

.

slipx06, 2024-05-06:
Yes you can scale the card using card_height, card_width and panel_mode. You would also need to adjust your dashboard view to the appropriate view type. I would suggest grid layout card

card_height: | Optional | 396px | Only used when panel_mode: false. Sets the card height in pixels. Specify the value i.e. 400px or provide a sensor i.e. input.numer_height
card_width: | Optional | 100% | Only used when panel_mode: true. Sets the card width in pixels or percentage 400px or 80%. For adjustments when using the Panel(1 card) view types or grid layouts

slipx06, 2024-05-12:
Continue here https://github.com/slipx06/sunsynk-power-flow-card/discussions/388

closed 2024-05-12 by slipx06