slipx06/sunsynk-power-flow-card #427: add Growatt SPF5000ES  support for status variables

tabascoz, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, I own some Growatt SPF5000ES and I noticed the code states in growatt.ts doesn't match neither general status err codes or PV codes for this model.

Would it be possible to include the status according to PV status variable the device generates, which to me it makes more sense? 

These are the values it generates (from my Homeassistant mqtt setup): 

    ` {% if (value_json['values']['pvstatus'] | int == 0) %}                                                                                                                                                       
        Waiting                                                                                                                                                                                                    
      {% elif (value_json['values']['pvstatus'] | int == 1) %}                                                                                                                                                     
        Normal                                                                                                                                                                                                     
      {% elif (value_json['values']['pvstatus'] | int == 2) %}                                                                                                                                                     
        Battery Discharging                                                                                                                                                                                        
      {% elif (value_json['values']['pvstatus'] | int == 3) %}                                                                                                                                                     
        Fault                                                                                                                                                                                                      
      {% elif (value_json['values']['pvstatus'] | int == 8) %}                                                                                                                                                     
        Hybrid Solar                                                                                                                                                                                               
      {% elif (value_json['values']['pvstatus'] | int == 9) %}                                                                                                                                                     
        MPPT Charge & Bypass                                                                                                                                                                                       
      {% elif (value_json['values']['pvstatus'] | int == 10) %}                                                                                                                                                    
        AC charge and Bypass                                                                                                                                                                                       
      {% elif (value_json['values']['pvstatus'] | int == 11) %}                                                                                                                                                    
        Grid Bypass                                                                                                                                                                                                
      {% elif (value_json['values']['pvstatus'] | int == 12) %}                                                                                                                                                    
        Solar/Battery                                                                                                                                                                                              
      {% else %}                                                                                                                                                                                                   
        Unknown                                                                                                                                                                                                    
      {% endif %} 
`

### Expected behaviour

Card shows Growatt SPF5000ES status according to its specs. 

### Mode

Both

### Context / Reason

SPF5000 family is a common inverter used in many solar installations.

slipx06, 2024-05-10:
Thanks I'll add that

slipx06, 2024-05-11:
Most of these are related to battery status. For now I've added pvstatus codes 0, 1 and 3 for standby, normal and fault status

```
    statusGroups: InverterStatus = {
        standby: {states: ['0', 'standby', 'stand-by'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['selftest', 'self-checking'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['1', 'normal', 'ok'], color: 'green', message: localize('common.normal')},
        alarm: {states: ['alarm'], color: 'orange', message: localize('common.alarm')},
        fault: {states: ['3', 'fault'], color: 'red', message: localize('common.fault')},
    }
```

slipx06, 2024-05-12:
Added to [4.37.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.37.0)

closed 2024-05-12 by slipx06

tabascoz, 2024-05-13:
Many thanks Sir! ;)

tabascoz, 2024-06-14:
Hi there slipx06, 

I tested the code, understand the status is more close to the battery, however the SPF5000es, being a hybrid inverter, is where all the energy management really happens. I changed the js to reflect at least key indicator of Off-grid and On-grid for some pvstatus variables and i feel it's better suitable for the display purpose. May i submit a PR ?

slipx06, 2024-06-15:
Go for it. Happy to take a look.