slipx06/sunsynk-power-flow-card #434: Problem mapping to string.

ikeare, 2024-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Russian language has a problem with text and elements overlapping
[image]

### Steps To Reproduce

Constantly

### Expected behaviour

Words should be wrapped on another line, probably

### Card Version

4.37.0

### Home Assistant Version

12.3

slipx06, 2024-05-13:
That's what happens when you use ChatGPT  😄 If you can help improve the translated text below I would be grateful 

```
{
    "common": {
        "description": "Карта Home Assistant для эмуляции потока энергии Sunsynk, отображаемого на экране инвертора",
        "version": "Версия",
        "runtime_to": "ВРЕМЯ ДО",
        "battery_floating": "ПЛАВАЮЩАЯ БАТАРЕЯ",
        "charge": "ЗАРЯД",
        "to": "ДО",
        "daily_load": "ЕЖЕДНЕВНАЯ НАГРУЗКА",
        "daily_aux": "ЕЖЕДНЕВНАЯ ВСПОМОГАТЕЛЬНАЯ",
        "daily_grid_buy": "ЕЖЕДНЕВНАЯ ПОКУПКА ИЗ СЕТИ",
        "daily_grid_sell": "ЕЖЕДНЕВНАЯ ПРОДАЖА В СЕТЬ",
        "daily_solar": "ЕЖЕДНЕВНАЯ СОЛНЕЧНАЯ ЭНЕРГИЯ",
        "daily_solar_left": "ЕЖЕДНЕВНАЯ СОЛНЕЧНАЯ ЭНЕРГИЯ / ОСТАЛОСЬ СЕГОДНЯ",
        "total_solar_generation": "ЕЖЕДНЕВНАЯ СОЛНЕЧНАЯ ЭНЕРГИЯ / TOTAL SOLAR",
        "daily_charge": "ЕЖЕДНЕВНАЯ ЗАРЯДКА",
        "daily_discharge": "ЕЖЕДНЕВНАЯ РАЗРЯДКА",
        "ratio": "Соотношение",
        "timer_off": "Таймер выключен",
        "timer_on": "Таймер включен",
        "priority_batt": "Приоритет батареи",
        "priority_load": "Приоритет нагрузки",
        "nonessential_name": "Не существенное",
        "aux_name": "Вспомогательное",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Существенное",
        "autarky": "Autarky",
        "discharging": "Разряд",
        "charging": "Зарядка",
        "waiting": "Ожидание",
        "standby": "Режим ожидания",
        "no_battery": "Без батареи",
        "selftest": "Самодиагностика",
        "normal": "Нормально",
        "alarm": "Авария",
        "fault": "Неисправность",
        "days": "дни",
        "hrs": "часов",
        "min": "мин",
        "flash": "Flash",
        "check": "Check",
        "idle": "Бездействие",
        "exporting": "Экспорт",
        "importing": "Импорт",
        "ongrid": "On-Grid",
        "offgrid": "Off-Grid",
        "normalstop": " Normal Stop",
        "shutdown": "Shutdown",
        "offline": "Offline",
        "running": "Running",
        "sleepmode": "Sleep Mode",
        "grid_name": "Grid",
        "limit": "Limit"
    },
    "errors": {
        "battery": {
            "bat": "Не определены атрибуты батареи",
            "energy": "Пожалуйста, включите атрибут батарейной энергии и значение в Втч, например, 5.32 Энергия батареи: 5320",
            "shutdown_soc": "Пожалуйста, включите атрибут батарейного shutdown_soc и значение, например, shutdown_soc: 20",
            "full_capacity": "Полная емкость должна быть в диапазоне от 80 до 100",
            "empty_capacity": "Пустая емкость должна быть <= 40",
            "show_daily": "Пожалуйста, включите атрибуты day_battery_charge_70 и day_battery_discharge_71 и идентификаторы сущностей"
        },
        "solar": {
[... 93 lines cut ...]
      "export_colour": "Export Colour",
      "nonessential_name": "Non Essential Name",
      "nonessential_icon": "Non Essential Icon",
      "energy_cost_decimals": "Energy Cost Decimals",
      "show_remaining_energy": "Show Remaining Energy",
      "cat_title": {
        "title": "Title Options",
        "general": "General Options",
        "inverter": "Inverter Options",
        "solar": "Solar Options",
        "battery": "Battery Options",
        "grid": "Grid Options",
        "load": "Load Options",
        "entities": "Entities",
        "aux": "Aux Options",
        "energy": "Energy Entities",
        "bat": "Battery Entities",
        "sol": "Solar Entities",
        "inv": "Inverter Entities",
        "ld": "Load Entities",
        "gri": "Grid Entities"
      }
    }
}
```

ikeare, 2024-05-15:
> That's what happens when you use ChatGPT 😄 If you can help improve the translated text below I would be grateful
> 
> ```
> {
>     "common": {
>         "description": "Карта Home Assistant для эмуляции потока энергии Sunsynk, отображаемого на экране инвертора",
>         "version": "Версия",
>         "runtime_to": "ВРЕМЯ ДО",
>         "battery_floating": "ПЛАВАЮЩАЯ БАТАРЕЯ",
>         "charge": "ЗАРЯД",
>         "to": "ДО",
>         "daily_load": "НАГРУЗКА",
>         "daily_aux": "ВСПОМОГАТЕЛЬНАЯ",
>         "daily_grid_buy": "ПОКУПКА ИЗ СЕТИ",
>         "daily_grid_sell": "ПРОДАЖА В СЕТЬ",
>         "daily_solar": "ЕЖЕДНЕВНАЯ СОЛНЕЧНАЯ ЭНЕРГИЯ",
>         "daily_solar_left": "СОЛНЕЧНАЯ ЭНЕРГИЯ / ОСТАЛОСЬ СЕГОДНЯ",
>         "total_solar_generation": "СОЛНЕЧНАЯ ЭНЕРГИЯ / ИТОГО СОЛНЕЧНАЯ ЭНЕРГИЯ",
>         "daily_charge": "ЗАРЯД",
>         "daily_discharge": "РАЗРЯД",
>         "ratio": "Соотношение",
>         "timer_off": "Таймер выключен",
>         "timer_on": "Таймер включен",
>         "priority_batt": "Приоритет батареи",
>         "priority_load": "Приоритет нагрузки",
>         "nonessential_name": "Не существенное",
>         "aux_name": "Вспомогательное",
>         "pv1_name": "PV1",
>         "pv2_name": "PV2",
>         "pv3_name": "PV3",
>         "pv4_name": "PV4",
>         "essential": "Существенное",
>         "autarky": "Autarky",
>         "discharging": "Разряд",
>         "charging": "Заряд",
>         "waiting": "Ожидание",
>         "standby": "Режим ожидания",
>         "no_battery": "Без батареи",
>         "selftest": "Самодиагностика",
>         "normal": "Нормально",
>         "alarm": "Авария",
>         "fault": "Неисправность",
>         "days": "дни",
>         "hrs": "часов",
>         "min": "мин",
>         "flash": "Flash",
>         "check": "Check",
>         "idle": "Бездействие",
>         "exporting": "Экспорт",
>         "importing": "Импорт",
>         "ongrid": "On-Grid",
>         "offgrid": "Off-Grid",
>         "normalstop": "Остановка",
>         "shutdown": "Отключение",
>         "offline": "Оффлайн",
>         "running": "Работает",
>         "sleepmode": "Спящий Режим",
>         "grid_name": "Сеть",
>         "limit": "Лимит"
>     },
>     "errors": {
>         "battery": {
>             "bat": "Не определены атрибуты батареи",
>             "energy": "Пожалуйста, включите атрибут батарейной энергии и значение в Втч, например, 5.32 Энергия батареи: 5320",
>             "shutdown_soc": "Пожалуйста, включите атрибут батарейного shutdown_soc и значение, например, shutdown_soc: 20",
>             "full_capacity": "Полная емкость должна быть в диапазоне от 80 до 100",
>             "empty_capacity": "Пустая емкость должна быть <= 40",
>             "show_daily": "Пожалуйста, включите атрибуты day_battery_charge_70 и day_battery_discharge_71 и идентификаторы сущностей"
>         },
>         "solar": {
[... 95 lines cut ...]
>       "nonessential_icon": "Иконка Не Обязательной Нагрузки",
>       "energy_cost_decimals": "Разрядность Стоимости Энергии",
>       "show_remaining_energy": "Показать Остаток Энергии",
>       "cat_title": {
>         "title": "Параметры Заголовка",
>         "general": "Общие Настройки",
>         "inverter": "Настройки Инвертора",
>         "solar": "Настройки Солнечного Поля",
>         "battery": "Настройки Батареи",
>         "grid": "Настройки Сети",
>         "load": "Настройки Нагрузки",
>         "entities": "Сущности",
>         "aux": "Настройки Всп. Нагрузки",
>         "energy": "Сущности Энергии",
>         "bat": "Сущности Батареи",
>         "sol": "Сущности Солнечного Поля",
>         "inv": "Сущности Инвертора",
>         "ld": "Сущности Нагрузки",
>         "gri": "Сущности Сети"
>       }
>     }
> }
> ```

How I see the card translation.

commit ee6f908a (2024-05-16): fix: add additional fields to the visual editor closes #433
locale: update Russian translation closes #434
docs: update documentation
closed 2024-05-16 by slipx06