slipx06/sunsynk-power-flow-card #436: Solar Max Power

devilliersjohnny, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi Again...

You closed the previous issue that I opened with regards to the input of sensors within the options of the card.

Sadly this issue is still not resolved using the latest version of your card... the work around ( using the code editor ) works for the battery energy and and battery shutdown_soc, however the solar max_power does not seem to accept this format?

Could you please look into this

P.S thank you for the great work so far

### Expected behaviour

A working front end UI input not the back end code version

### Card Version

4.37.1

### Home Assistant Version

2024.5.4

slipx06, 2024-05-18:
Hi. The latest update allows you to configure a sensor for battery energy and and battery shutdown_soc. If you look at the screenshot below there are two new input fields on the right that allow you to select entities.  Alternatively you can use the numeric selectors on the left. The two are mutually exclusive. 

[image]

Solar max_power does not currently support sensor input (unlike battery max_power) but I can add that.

slipx06, 2024-05-18:
To make this a bit clearer in the visual editor I'm adding an expandable `optional sensors` section to each of the card sections which can be used to provide sensors as an alternative to entering values in the options above. If you've provided a sensor the input field above will be empty as the selector is expecting a numeric value etc. If you provide a value above it pulls this through to the sensor field below. 

[image]

slipx06, 2024-05-19:
@devilliersjohnny this has been implemented in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.37.2

devilliersjohnny, 2024-05-19:
@slipx06 THANK YOU! For indulging me... so far!!! For now I still have one more question? About the solar_max?

slipx06, 2024-05-19:
What do you mean be solar_max?

devilliersjohnny, 2024-05-19:
Apologies should have been clearer, the ability to implement a sensor within the solar_max fields?

slipx06, 2024-05-19:
Do you mean

```
  pv1_max_power: 
  pv2_max_power: 
  pv3_max_power:
  pv4_max_power:
```

devilliersjohnny, 2024-05-19:
Apologies for the rough edit but either the max_power total or the 4 pv_max powers that add up please! 
 
[image]

slipx06, 2024-05-19:
You can already provide a sensor for max_power. I can add the others.

[image]
[image]

devilliersjohnny, 2024-05-20:
HI @slipx06 thank you can confirm that the max power appears to be working, I am currently using a template to calculate every thing for my end users...

```yaml
  - name: "EDB I1 MPPT1 Max Power"
    unique_id: edb_i1_mppt1_max_power
    unit_of_measurement: "W"
    icon: mdi:solar-power
    device_class: power
    state: >
      {% if states('input_number.edb_inverter_count_input_number') | float >= 1
         and states('input_select.edb_i1_config_has_solar') == "True"
         and states('input_number.edb_i1_mppt_count_input_number') | float >= 1
         and states('input_text.edb_i1_mppt1_panel_size_input_text') | is_number
         and states('input_text.edb_i1_mppt1_panel_amount_input_text') | is_number %}
          {% set panel_size = states('input_text.edb_i1_mppt1_panel_size_input_text')  | float | round(0) %}
          {% set panel_count = states('input_text.edb_i1_mppt1_panel_amount_input_text')  | float | round(0) %}
          {% set mppt_max_power = panel_size * panel_count %}
        {% else %}
          {% set mppt_max_power = 'None' %}
      {% endif %}
      {{ mppt_max_power }}
```

would be nice if I could have the animation that you include ( the partial coloration ) working with my setup as well. 

Again THANK YOU for all the work done so far!

commit a84d4c08 (2024-05-22): feat: pv max power attributes accept sensors closes #436
fix: Victron status codes closes #438
locale: update Italian translation closes #424
closed 2024-05-22 by slipx06