slipx06/sunsynk-power-flow-card #437: More then 100%

sitvesep, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

If the value on the string setting exceed  the maximum value set, then it is shown still 100% even if under certain conditions it can produce more energy from the string. 
[image]

### Expected behaviour

See more than 100% when the value is exceeded

### Possible Solutions

add the option to show more than 100% when the value is exceeded

### Mode

Both

### Context / Reason

under certain conditions, photovoltaics can produce much more than the nominal value for a short period of time. it would be nice to see that is for example 115%

Thank you

slipx06, 2024-05-18:
I will add that for you

sitvesep, 2024-05-18:
Thanks a lot for your great work and support.

closed 2024-05-19 by slipx06

matomatusov, 2024-05-19:
@sitvesep 
Hi, how did you get this there?
[image]

Thanks

sitvesep, 2024-05-19:
> @sitvesep Hi, how did you get this there? ![1](https://private-user-images.githubusercontent.com/129114957/331883627-87f6d17f-be4b-408b-b786-fa9ba46e5712.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzQxMzUsIm5iZiI6MTcxNjEzMzgzNSwicGF0aCI6Ii8xMjkxMTQ5NTcvMzMxODgzNjI3LTg3ZjZkMTdmLWJlNGItNDA4Yi1iNzg2LWZhOWJhNDZlNTcxMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxOVQxNTUwMzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNWNkMjc2ZGFiMTQ4YjkwODBmYjNmMGIwMDUzYjBlNzdkMTMxN2VjMGNiMzdhODEzMTZlYTQ2NWQwODM4YjEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.CuwqT2PUf0AtSvtbaB58ZamN3CsRP7GTBzJ1fdtyr8k)
> 
> Thanks

I have it filled in like this:

[image]
[image]
[image]
[image]

matomatusov, 2024-05-19:
I have a similar setup but it doesn't show up there. That's the minimum SOC setting?

sitvesep, 2024-05-19:
type:%20custom%3Asunsynk-power-flow-card%0Acardstyle%3A%20lite%0Apanel_mode%3A%20true%0Acard_width%3A%20100%25%0Ashow_solar%3A%20true%0Atitle%3A%20FVE%207,2%20kWp%0Atitle_colour%3A%20grey%0Atitle_size%3A%2015px%0Ashow_battery%3A%20true%0Alarge%20font%3A%20true%0Ainverter%3A%0A%20%20model%3A%20goodwe_gridmode%0A%20%20modern%3A%20null%0A%20%20colour%3A%20white%0A%20%20autarky%3A%20energy%0A%20%20three_phase%3A%20true%0A%20%20animation_speed%3A%205%0Abattery%3A%0A%20%20energy%3A%2014200%0A%20%20shutdown_soc%3A%2010%0A%20%20invert_power%3A%20false%0A%20%20colour%3A%0A%20%20%20%20-%20235%0A%20%20%20%20-%2077%0A%20%20%20%20-%2061%0A%20%20show_daily%3A%20true%0A%20%20max_power%3A%208800%0A%20%20full_capacity%3A%20100%0A%20%20empty_capacity%3A%2020%0A%20%20show_absolute%3A%20false%0A%20%20auto_scale%3A%20true%0A%20%20animation_speed%3A%203%0A%20%20charge_colour%3A%0A%20%20%20%20-%200%0A%20%20%20%20-%20157%0A%20%20%20%20-%2014%0A%20%20dynamic_colour%3A%20true%0A%20%20show_remaining_energy%3A%20true%0A%20%20linear_gradient%3A%20true%0A%20%20hide_soc%3A%20false%0Asolar%3A%0A%20%20colour%3A%0A%20%20%20%20-%20255%0A%20%20%20%20-%20255%0A%20%20%20%20-%2044%0A%20%20show_daily%3A%20true%0A%20%20display_mode%3A%202%0A%20%20mppts%3A%202%0A%20%20animation_speed%3A%203%0A%20%20max_power%3A%207200%0A%20%20pv1_name%3A%20PV1%0A%20%20pv2_name%3A%20PV2%0A%20%20dynamic_colour%3A%20true%0A%20%20pv1_max_power%3A%203600%0A%20%20pv2_max_power%3A%203600%0A%20%20auto_scale%3A%20true%0Aload%3A%0A%20%20show_daily%3A%20true%0A%20%20dynamic_colour%3A%20true%0A%20%20load1_name%3A%20APC/Router/NAS%0A%20%20load2_name:%20ARUBA%0A%20%20additional_loads:%203%0A%20%20dynamic_icon:%20true%0A%20%20invert_load:%20false%0A%20%20auto_scale:%20false%0A%20%20load3_name:%20Mycka%0A%20%20animation_speed:%202%0A%20%20max_power:%2017250%0A%20%20essential_name:%20Dum%0A%20%20colour:%0A%20%20%20%20-%20102%0A%20%20%20%20-%20219%0A%20%20%20%20-%20223%0A%20%20load1_icon:%20mdi:nas%0A%20%20load2_icon:%20mdi:wifi%0A%20%20load3_icon:%20mdi:dishwasher%0Agrid:%0A%20%20show_daily_buy:%20false%0A%20%20show_daily_sell:%20false%0A%20%20show_nonessential:%20true%0A%20%20invert_grid:%20true%0A%20%20animation_speed:%203%0A%20%20export_colour:%0A%20%20%20%20-%200%0A%20%20%20%20-%20255%0A%20%20%20%20-%200%0A%20%20colour:%0A%20%20%20%20-%20255%0A%20%20%20%20-%200%0A%20%20%20%20-%200%0A%20%20grid_name:%20EGD%0Aentities:%0A%20%20use_timer_248:%20switch.bojlery%0A%20%20day_battery_charge_70:%20sensor.today_battery_charge%0A%20%20day_battery_discharge_71:%20sensor.today_battery_discharge%0A%20%20day_load_energy_84:%20sensor.today_load%0A%20%20day_pv_energy_108:%20sensor.today_s_pv_generation%0A%20%20inverter_voltage_154:%20sensor.on_grid_l1_voltage%0A%20%20inverter_voltage_L2:%20sensor.on_grid_l2_voltage%0A%20%20inverter_voltage_L3:%20sensor.on_grid_l3_voltage%0A%20%20load_frequency_192:%20sensor.meter_frequency%0A%20%20inverter_current_164:%20sensor.on_grid_l1_current%0A%20%20inverter_current_L2:%20sensor.on_grid_l2_current%0A%20%20inverter_current_L3:%20sensor.on_grid_l3_current%0A%20%20inverter_power_175:%20zero%0A%20%20grid_power_167:%20sensor.house_consumption%0A%20%20pv1_power_186:%20sensor.pv1_power%0A%20%20battery_voltage_183:%20sensor.battery_voltage%0A%20%20battery_soc_184:%20sensor.battery_state_of_charge%0A%20%20battery_power_190:%20sensor.battery_power%0A%20%20battery_current_191:%20sensor.battery_current%0A%20%20load_power_L1:%20sensor.load_l1%0A%20%20load_power_L2:%20sensor.load_l2%0A%20%20load_power_L3:%20sensor.load_l3%0A%20%20grid_ct_power_total:%20sensor.active_power%0A%20%20grid_ct_power_172:%20sensor.meter_active_power_l1%0A%20%20grid_ct_power_L1:%20sensor.meter_active_power_l1%0A%20%20grid_ct_power_L2:%20sensor.meter_active_power_l2%0A%20%20grid_ct_power_L3:%20sensor.meter_active_power_l3%0A%20%20grid_total:%20sensor.load%0A%20%20pv1_voltage_109:%20sensor.pv1_voltage%0A%20%20pv1_current_110:%20sensor.pv1_current%0A%20%20pv2_voltage_111:%20sensor.pv2_voltage%0A%20%20pv2_current_112:%20sensor.pv2_current%0A%20%20pv_total:%20sensor.pv_power%0A%20%20grid_connected_status_194:%20sensor.grid_mode_code%0A%20%20inverter_status_59:%20sensor.work_mode_code%0A%20%20battery_temp_182:%20sensor.battery_temperature%0A%20%20radiator_temp_91:%20sensor.inverter_temperature_radiator%0A%20%20dc_transformer_temp_90:%20sensor.inverter_temperature_air%0A%20%20battery_status:%20sensor.battery_mode_code%0A%20%20remaining_solar:%20sensor.solcast_pv_forecast_forecast_remaining_today%0A%20%20essentil_power:%20sensor.house_consumption%0A%20%20essential_load1:%20sensor.shellyplug_s_80646f811e55_power%0A%20%20essential_load2:%20sensor.shellyplug_s_80646f822703_power%0A%20%20essential_load3:%20sensor.shellyplusplugs_d4d4daeb7d98_switch_0_power%0A%20%20environment_temp:%20sensor.shellyht_7dd21d_temperature%0A%20%20pv2_power_187:%20sensor.pv2_power%0Amax_line_width:%208%0Ashow_grid:%20true%0Adynamic_line_width:%20true%0A

sitvesep, 2024-05-19:
try to compare directly in Yamal. I think I wrote something directly according to the documentation.

[image]

lamixer, 2024-05-19:
Thanks! I wanted this too!

commit 0e4644ed (2024-05-19): fix: solar eficiency can exceed 100% (limited to 200%) closes #437
fix: update visual editor.