slipx06/sunsynk-power-flow-card #438: Victron status incorrect

drtinaz, 2024-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I have selected the vebus state entity in the entities list. On the overview the inverter status just shows 'status'. I have looked in the .js file and the states that are included for victron (numeric and textual both) do not correspond to the correct values provided by victron dbus. For example, vebus state while inverting is '9'. The possible states values from dbus are 0 thru 11 and 252. (I'm using mqtt for my Victron integration). This is with a Multiplus 2.

Here is part of the yaml for the vebus state sensor with correct values

```
{% if value_json.value == 0 %}
                Off
            {% elif value_json.value == 1 %}
                Low Power
            {% elif value_json.value == 2 %}
                Fault
            {% elif value_json.value == 3 %}
                Bulk
            {% elif value_json.value == 4 %}
                Absorption
            {% elif value_json.value == 5 %}
                Float
            {% elif value_json.value == 6 %}
                Storage
            {% elif value_json.value == 7 %}
                Equalize
            {% elif value_json.value == 8 %}
                Passthru
            {% elif value_json.value == 9 %}
                Inverting
            {% elif value_json.value == 10 %}
                Power Assist
            {% elif value_json.value == 11 %}
                Power Supply
            {% elif value_json.value == 252 %}
                External Control
```

### Expected behaviour

I'm hoping this can be updated in a future release

### Card Version

v4.37.2

### Home Assistant Version

Core 2024.5.4 Supervisor 2024.05.1 Operating System 12.3 Frontend 20240501.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 14336
  shutdown_soc: 10
  show_daily: false
  shutdown_soc_offgrid: 10
  auto_scale: true
  dynamic_colour: true
  max_power: 5120
  show_remaining_energy: true
  animate: true
  linear_gradient: true
  invert_power: true
  show_absolute: true
solar:
  show_daily: true
  mppts: 2
  dynamic_colour: true
  auto_scale: true
  pv1_name: Front
  pv2_name: Rear
  max_power: 1600
  pv1_max_power: 1000
  pv2_max_power: 600
  efficiency: 1
  display_mode: 1
load:
  show_daily: false
  auto_scale: true
  dynamic_icon: true
  dynamic_colour: true
  additional_loads: 2
  show_aux: true
  aux_dynamic_colour: true
```

drtinaz, 2024-05-19:
sorry somehow I didn't get the entire config the first time...

```
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 14336
  shutdown_soc: 10
  show_daily: false
  shutdown_soc_offgrid: 10
  auto_scale: true
  dynamic_colour: true
  max_power: 5120
  show_remaining_energy: true
  animate: true
  linear_gradient: true
  invert_power: true
  show_absolute: true
solar:
  show_daily: true
  mppts: 2
  dynamic_colour: true
  auto_scale: true
  pv1_name: Front
  pv2_name: Rear
  max_power: 1600
  pv1_max_power: 1000
  pv2_max_power: 600
  efficiency: 1
  display_mode: 1
load:
  show_daily: false
  auto_scale: true
  dynamic_icon: true
  dynamic_colour: true
  additional_loads: 2
  show_aux: true
  aux_dynamic_colour: true
  aux_loads: 1
  load1_name: HP1
  load2_name: HP2
  load3_name: HP3
  aux_name: DC/Alt
  aux_load1_name: DC/Alt
  essential_name: AC Loads
  show_absolute_aux: true
  invert_aux: false
  max_power: 12000
  load1_icon: ''
  load2_icon: ''
  load3_icon: ''
  show_daily_aux: false
  invert_load: false
  aux_type: mdi:compare-horizontal
grid:
  show_daily_buy: false
  show_daily_sell: false
  show_nonessential: false
  auto_scale: true
  show_absolute: true
  additional_loads: 0
  max_power: 12000
entities:
  inverter_voltage_154: sensor.multiplus_l1_output_voltage
  load_frequency_192: sensor.multiplus_l1_output_frequency
  inverter_current_164: sensor.multiplus_l1_output_current
  inverter_power_175: sensor.inverted_output
  battery_voltage_183: sensor.smart_shunt_500_main_voltage
  battery_soc_184: sensor.smart_shunt_500_main_soc
  battery_power_190: sensor.smart_shunt_500_main_power
  battery_current_191: sensor.smart_shunt_500_main_current
  grid_power_169: sensor.multiplus_total_input_power
  grid_ct_power_172: sensor.multiplus_l1_input_power
  aux_power_166: sensor.dc_systems_dc_systems_power
  pv1_power_186: sensor.mppt_100_50_mppt100_50_pv_power
  pv2_power_187: sensor.mppt_100_30_mppt100_30_pv_power
  pv1_voltage_109: sensor.mppt_100_50_mppt100_50_pv_voltage
  pv1_current_110: sensor.mppt_100_50_mppt100_50
  pv2_voltage_111: sensor.mppt_100_30_mppt100_30_pv_voltage
  pv2_current_112: sensor.mppt_100_30_mppt100_30
  battery_temp_182: sensor.daly_daly_daly_temperatur
  inverter_voltage_L2: sensor.multiplus_l2_output_voltage
  inverter_current_L2: sensor.multiplus_l2_output_current
  essential_load1: sensor.kauf_plug_power
  essential_load2: sensor.kauf_plug_power_4
  essential_load3: sensor.smart_shunt_500_main_power
  load_power_L1: sensor.kauf_plug_power
  load_power_L2: sensor.kauf_plug_power_4
  load_power_L3: sensor.shelly_shellyplus1pm_30c922825e44_current_consumption
  grid_ct_power_L2: sensor.multiplus_l2_input_power
  grid_voltage: sensor.multiplus_l1_input_voltage
  aux_load1: sensor.dc_systems_dc_systems_power
  aux_load1_extra: sensor.dc_systems_dc_systems_current
  essential_load1_extra: sensor.kauf_plug_current
  essential_load2_extra: sensor.kauf_plug_current_4
  day_pv_energy_108: sensor.combined_solar_yield_today
  essential_power: sensor.combined_l1_l2_output
  environment_temp: sensor.cerbo_gx_outdoor_temp
  radiator_temp_91: sensor.cerbo_gx_inv_bay_temp
  inverter_status_59: sensor.multiplus_vebus_state
show_battery: true
show_grid: true
dynamic_line_width: true
panel_mode: false
inverter:
  three_phase: true
  auto_scale: true
  model: victron
  modern: false
  autarky: 'no'
large_font: true

```

Pho3niX90, 2024-05-20:
> I'm certainly more than willing to update the victron.ts file if only I was able to find it on my system. I'm happy to do the work and test it on my own system, I just don't know where the file is located in HA.

you won't find it, all files are compiled into a js file. To test it, you will need to clone the repo, make the changes and compile the js

Pho3niX90, 2024-05-20:
This is the victron specific config. https://github.com/slipx06/sunsynk-power-flow-card/blob/main/src/inverters/brands/victron.ts

The question at hand, is the status wrong for all victrons, or only this specific model? `Multiplus 2`

If only the model, we probably have to take that into consideration and add model specific configs.

drtinaz, 2024-05-20:
> This is the victron specific config. https://github.com/slipx06/sunsynk-power-flow-card/blob/main/src/inverters/brands/victron.ts
>
> The question at hand, is the status wrong for all victrons, or only [...]

The status values are going to be the same with the vast majority of victron models, if not all. If there are any outliers, they would most likely be much older more basic units. Victron tries to use the same registers and value representations across all platforms as much as is possible because of Venus os (Victron's software for the GX control devices).

I am familiar with GitHub and forking/cloning etc, and would certainly be happy to make the changes and do some testing. However I wouldn't know how to actually compile the .js after making the changes. I have built packages for victron Venus os but that doesn't require any compiling. I use GitHub desktop and virtual studio and I know almost nothing when it comes to command line and GitHub.

drtinaz, 2024-05-20:
I don't know what I'm doing wrong but I am not able to install yarn on my dev system. I've spent hours just trying to figure that part out. I can't get past it.

drtinaz, 2024-05-21:
Ok I finally got it all figured out and got yarn working properly on my dev PC. I added/updated the victron status codes and manually installed the new package on my HA. The status dot is displaying now and changes colors correctly according to the current status, but there is no text being displayed. I have it in 'full' mode. Is there something else I need to do for it to display the status text mode in addition to the dot? 

Once I have this figured out and verified all is well I can issue a PR back to main.

drtinaz, 2024-05-22:
finally figured it out. I had to add a few new states to the common file. Now the status shows correct dot and text. I will now issue a PR.

commit a84d4c08 (2024-05-22): feat: pv max power attributes accept sensors closes #436
fix: Victron status codes closes #438
locale: update Italian translation closes #424
closed 2024-05-22 by slipx06