slipx06/sunsynk-power-flow-card #446: question: what is this SOC?

matomatusov, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi,
what is this SOC? I don't see it there and I don't know why.
[image]

Thanks

### Expected behaviour

what is this SOC? I don't see it there and I don't know why.

### Card Version

4.37.2

### Home Assistant Version

2024.5.4

wacalitz, 2024-05-21:
Under Battery Options 
I believe it is the Shutdown SOC you've entered in the card
Lower down there is a toggle Hide SOC where you can toggle to show it or not.

Mine is set to off as I have markdown card giving me the relative SOC targets when charging and discharging.

matomatusov, 2024-05-21:
@wacalitz 

That Hide SOC does nothing for me.
It's always the same.

slipx06, 2024-05-26:
It displays the SOC setting (capacity) for the selected program mode/ timer mode. You need to provide a `use_timer_248:` sensor as well as program sensors below
```
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_prog1_capacity
  prog1_charge: switch.sunsynk_prog1_grid_charge
  prog2_time: sensor.sunsynk_time_slot_2
  prog2_capacity: number.sunsynk_prog2_capacity
  prog2_charge: switch.sunsynk_prog2_grid_charge
  prog3_time: sensor.sunsynk_time_slot_3
  prog3_capacity: number.sunsynk_prog3_capacity
  prog3_charge: switch.sunsynk_prog3_grid_charge
  prog4_time: sensor.sunsynk_time_slot_4
  prog4_capacity: number.sunsynk_prog4_capacity
  prog4_charge: switch.sunsynk_prog4_grid_charge
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.sunsynk_prog5_capacity
  prog5_charge: switch.sunsynk_prog5_grid_charge
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.sunsynk_prog6_capacity
  prog6_charge: switch.sunsynk_prog6_grid_charge
```
Alternatively if the selected inverter is Goodwe or Huawei it will show the shutdown soc and the shutdown soc off_grid. 
Its not needed if you dont make use of the program mode as the shutdown soc is shown in the text

[image]

commit 55d1874b (2024-05-27): fix: alignment of energy cost buy and sell
fix: display shutdown soc and shutdown off grid soc for all inverters if program mode or system timer is false. Use `hide_soc: true` to hide closes #446
docs: update configuration
closed 2024-05-27 by slipx06