slipx06/sunsynk-power-flow-card #449: color the flow

jolly12f, 2024-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

at this moment the flow that goes from the inverter towards the house remains blue

### Expected behaviour

I would like to color the stream based on the source it is using

### Possible Solutions

It should take the color of the battery, the sun or the grid, so it would be nice to see the flow of the same color going towards the load. the house icon is already colored that way, I don't think there are any problems coloring the flow too

### Mode

Both

### Context / Reason

it would be more beautiful to look at

jolly12f, 2024-05-25:
[image]

slipx06, 2024-05-26:
Hi. Something like this?

[image]

Roving-Ronin, 2024-05-26:
@slipx06 dumb question... whats the 'Limit: 8.00 kW' map to ? 
I'm assuming it shows if there's a limit on the 'maximum export = xxxx W'?
Is this able to be linked to a sensor, as Huawei/iStore with WLCRS integration v1.4.0 (Alpha atm) adds:

- That sets the maximum export in Watts to the Grid:
 service: huawei_solar.set_maximum_feed_grid_power

- That sets the maximum export in % of overall inverter(s) capacity, to the Grid:
 service: huawei_solar.set_maximum_feed_grid_power_percentage   

- That allows resetting the maximum grid feed back to 100% of capabilities:
 service: huawei_solar.reset_maximum_feed_grid_power 

IIRC this is then sent to the inverters (or primary only that controls all of them?) at:
number.inverter_1_power_derating  or number.inverter_1_power_derating_by_percentage

slipx06, 2024-05-26:
Yes maximum export to grid based on Watts. You can provide a sensor e.g. `max_sell_power: huawei_solar.set_maximum_feed_grid_power`

jolly12f, 2024-05-26:
> CIAO. Qualcosa come questo?
> 
> ![Immagine](https://private-user-images.githubusercontent.com/7227275/333871793-6e18a2db-0b87-4ca1-977e-c171579da2a2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3MjgyNTksIm5iZiI6MTcxNjcyNzk1OSwicGF0aCI6Ii83MjI3Mjc1LzMzMzg3MTc5My02ZTE4YTJkYi0wYjg3LTRjYTEtOTc3ZS1jMTcxNTc5ZGEyYTIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjZUMTI1MjM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjhhYjcwOTBhNzA5M2I0MmE0ZjdkNDQ1ZGM5MmNmZDczYmJhNzI4YWJiNDM2MjIwZTJhNGJmODBiNmUxY2Y5YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pBVrw8uQVjyRUzNZhaw6oKn2fApNTrEgAp2KEGiG-JI)

yes exactly, I would also color the tables with numbers!!

slipx06, 2024-05-26:
I've only applied this to the path element for now and also implemented for the battery

closed 2024-05-26 by slipx06

jolly12f, 2024-05-27:
> Per ora l'ho applicato solo all'elemento percorso e l'ho implementato anche per la batteria

I just updated but this doesn't happen
[image]

Roving-Ronin, 2024-05-27:
@slipx06  thanks for this update. Could you please check your HA and see if you get this error also? I have the following sensors, like this (my own HA has a more complex tariff plan, but same format and issue):

```
    - name: "Electricity - Price" 
      unique_id: electricity_price
      unit_of_measurement: "$/kWh"
      device_class: monetary
      state: >
        {## Enter cost per kWh rates below. Rate 1 ##}
        {% set rate1 = 0.30812 %}
        {% set rate2 = 0.0 %}
        {% set power_meter_active_power = states('sensor.power_meter_active_power') | float %}
        {% if power_meter_active_power <= 0 %} {{rate1}}
        {% else %} {{rate2}}
        {% endif %}

#  Current FIT rate, rate 1 & 2 allows setting a second rate if variable FIT rates (i.e. $0.15 for first 10kWh, then $0.07 thereafter)

    - name: "Electricity - FIT"
      unique_id: electricity_fit
      unit_of_measurement: "$/kWh"
      device_class: monetary
      icon: mdi:cash-plus
      state: >
        {## Enter compensation per kWh rates below ##}
        {% set rate1 = 0.07 %}
        {% set rate2 = 0.07 %}
        {% set fit_exported_today = states('sensor.hs_grid_exported_daily') | float %}
        {% if fit_exported_today <= 10 %} {{rate1}}
        {% else %} {{rate2}}
        {% endif %}
```

These sensors output a Grid Export price (FIT) that when over 10kWh exported for the day reads as 0.07  (device_class: monetary so in the dash shows with the $), conversely the above it setup to say if the Power Meter is reading as <=0 Watts i.e. exporting, then report 0.30812

Given this when exporting the price showing next to the grid should be $0.07 $/kWh, however its showing as 0.07} $/kWh. 

...change of plans, I think I managed to find the typo myself. I've lodged PULL request if you want to check it please ?

https://github.com/slipx06/sunsynk-power-flow-card/pull/457

Roving-Ronin, 2024-05-27:
@slipx06 just noticed this, now that I'm importing at night. Bit of overlap on the 'h' from kWh and the transmission tower.

[image]

slipx06, 2024-05-27:
> @slipx06 just noticed this, now that I'm importing at night. Bit of overlap on the 'h' from kWh and the transmission tower. [...]

Thanks I've corrected this

github-daniel-stoian, 2024-05-28:
@slipx06 It's not working on my side. The load line is still blue.

- type: custom:sunsynk-power-flow-card
    cardstyle: compact
    panel_mode: true
    show_solar: true
    dynamic_line_width: false
    max_line_width: 8
    min_line_width: 2
    inverter:
      modern: false
      colour: '#1F618D'
      model: sofar
      autarky: 'no'
      three_phase: false
    battery:
      energy: 3500
      shutdown_soc: 20
      show_daily: false
      colour: '#47d147'
      max_power: 3500
      full_capacity: 100
      empty_capacity: 30
      dynamic_colour: true
      linear_gradient: true
    solar:
      show_daily: true
      mppts: 2
      colour: '#ffa31a'
      max_power: 6000
      pv1_name: PV1 Vest
      pv1_max_power: 4200
      pv2_name: PV2 Est
      pv2_max_power: 3500
      efficiency: 3
    load:
      show_daily: true
      dynamic_colour: true
      dynamic_icon: false
      colour: DeepSkyBlue
      max_power: 6000
      additional_loads: 4
      load1_name: Boiler
      load2_name: Heaters
      load3_name: DP1
      load4_name: DP2
      load1_icon: mdi:water-boiler
      load2_icon: mdi:heat-pump
      load3_icon: mdi:generator-portable
      load4_icon: mdi:generator-portable
    grid:
      show_daily_buy: true
      show_daily_sell: false
      show_nonessential: true
      max_power: 6000
      colour: '#ff0000'
[image]

slipx06, 2024-05-28:
@github-daniel-stoian I think I found the issue. Can you enable dynamic Icon under load options and see if it enables this?

github-daniel-stoian, 2024-05-28:
@slipx06 No, it does not work.
I've played extensivelly with dynamic_icon and dynamic_colour setting them true or false in any possible combination.

[image]

slipx06, 2024-05-28:
No that's expected behavior. The colour of the line will only change when a single source is able to supply 100% of the load. Wait until you have enough PV and it will change. You have to have `dynamic_icon: true` but I'll fix that

github-daniel-stoian, 2024-05-28:
It does not work when PV is able to supply 100% of the load.

But somehow it does work when Grid is providing 100% of the load.

[image]

[image]

slipx06, 2024-05-28:
Are you sure you have dynamic_icon is set to true in your first screenshot. The icon should show the picture of the house with solar panles and sun when the PV supply exceeds the load? In any case I need to update the code to remove the dependency on this setting. So wait for the next release to test.

jolly12f, 2024-05-28:
in my case it doesn't color when 100% comes from the grid

github-daniel-stoian, 2024-05-28:
Are you sure you have dynamic_icon is set to true in your first screenshot: Yes, I'm sure.
The icon should show the picture of the house with solar panles and sun when the PV supply exceeds the load? : No

[image]

slipx06, 2024-05-28:
Can you share your card config. Do you have a sensor defined for non essential power and aux power?

github-daniel-stoian, 2024-05-28:
No sensors for non-essential and aux power.

- type: custom:sunsynk-power-flow-card
    cardstyle: compact
    panel_mode: true
    show_solar: true
    dynamic_line_width: false
    max_line_width: 8
    min_line_width: 2
    inverter:
      modern: false
      colour: '#1F618D'
      model: sofar
      autarky: 'no'
      three_phase: false
    battery:
      energy: 3500
      shutdown_soc: 20
      show_daily: false
      colour: '#47d147'
      max_power: 3500
      full_capacity: 100
      empty_capacity: 30
      dynamic_colour: true
      linear_gradient: true
    solar:
      show_daily: true
      mppts: 2
      colour: '#ffa31a'
      max_power: 6000
      pv1_name: PV1 Vest
      pv1_max_power: 4200
      pv2_name: PV2 Est
      pv2_max_power: 3500
      efficiency: 3
    load:
      show_daily: true
      dynamic_colour: true
      dynamic_icon: true
      colour: DeepSkyBlue
      max_power: 6000
      additional_loads: 4
      load1_name: Boiler
      load2_name: Heaters
      load3_name: DP1
      load4_name: DP2
      load1_icon: mdi:water-boiler
      load2_icon: mdi:heat-pump
      load3_icon: mdi:generator-portable
      load4_icon: mdi:generator-portable
    grid:
      show_daily_buy: true
      show_daily_sell: false
      show_nonessential: true
      max_power: 6000
      colour: '#ff0000'
    entities:
      battery_voltage_183: sensor.uyn2hxn009_battery_voltage
      battery_current_191: sensor.uyn2hxn009_battery_ac_charge_current
      battery_soc_184: sensor.uyn2hxn009_state_of_charge
      battery_power_190: sensor.uyn2hxn009_battery_power
      day_battery_charge_70: sensor.uyn2hxn009_battery_ac_charge_energy_today
      day_battery_discharge_71: sensor.uyn2hxn009_battery_discharged_energy_today
      grid_power_169: sensor.uyn2hxn009_ac_discharge_load_consumption_power
      grid_voltage: sensor.uyn2hxn009_grid_voltage
      day_grid_import_76: sensor.uyn2hxn009_ac_discharge_energy_ac_bypass_today
      grid_ct_power_172: sensor.grid_discharge_power
      day_load_energy_84: sensor.consumed
      day_pv_energy_108: sensor.uyn2hxn009_power_generation_today_pvenergytoday
      pv1_power_186: sensor.uyn2hxn009_pv1_power
      pv2_power_187: sensor.uyn2hxn009_pv2_power
      pv1_voltage_109: sensor.pv1_voltage
      inverter_current_164: sensor.uyn2hxn009_inverter_current
      inverter_voltage_154: sensor.uyn2hxn009_inverter_output_voltage
      pv2_voltage_111: sensor.pv2_voltage
      essential_load1: sensor.boiler_power
      essential_load2: sensor.heaters_consumption
      essential_load3: sensor.dp1_battery_level
      essential_load4: sensor.dp2_battery_level
      essential_load1_extra: sensor.dp1_solar_in_power
      essential_load2_extra: sensor.dp2_solar_in_power

github-daniel-stoian, 2024-06-01:
Seems to work now after I've changed the followings:
```
    solar:
      off_threshold: 1
      
   load:
      off_threshold: 1
      path_threshold: 90
      
    grid:
      no_grid_colour: grey
      off_threshold: 1
      
     entities:
      grid_power_169 replaces with essential_power
```

[image]

commit fb10aebc (2024-05-26): feat: apply dynamic colour for load and battery to the flow paths closes #449
fix: handle wh, kwh and mwh for battery energy sonsor
fix: alignment of non_essential load 1 and load 2 extra information
fix: alignment of energy_cost_sell and energy_cost_buy data
fix  alignment of daily load on full card