slipx06/sunsynk-power-flow-card #450: Back Button Android

floschad, 2024-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When I push back button after I clicked a value to see details nothing happens, 2nd time App will close.

### Expected behaviour

Normally when you push back button it will close the detail window in the App.

### Card Version

4.38.0

### Home Assistant Version

2024.5.5

slipx06, 2024-05-27:
Thanks I can confirm this behavior.  Will try figure out how to fix.

Pho3niX90, 2024-05-27:
It seems I cannot reproduce this, which android versions are you using, and which devices? @slipx06

Pho3niX90, 2024-05-27:
After upgrading to 24.5.5, I experience the same. Will quickly test.

slipx06, 2024-05-27:
Not just Android. I can replicate in Chrome on Windows.

PR #459 (2024-05-31): fix: editor restructure
fixes: #450
closed 2024-05-31 by slipx06