slipx06/sunsynk-power-flow-card #456: Bug - v4.39.0 - Batteries Rated Capacity Duplication

Roving-Ronin, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In the visual editor there is two locations that the batteries rated capacity can be set using a sensor:

1) Battery Options ---> Optional Sensors (where a sensor can be used instead of a hard coded number) ---> Energy.  That in the yaml is stored as:

```
battery:
  energy: sensor.batteries.rated_capacity
```

2) In Entities ---> Battery Entities ---> battery_rated_capacity (that allows selecting a sensor to provide the value also). This is stored in the yaml as:

```
entities:
  battery_rated_capacity: sensor.batteries_rated_capacity
```

I've found that if both of these options are filled in, even with the same sensor, it means you end up with two locations specifying essentially the same thing. Checking the documentation it states "_Battery rated capacity (Ah). If provided this sensor will be used to calculate battery energy. The energy attribute under the battery card configuration will be ignored._".
However instead of ignoring the first entry (energy) what is happening is that the batteries capacity (or charge in kWh held at that moment) is being halved (i.e. 10kWh is shown as 5kWh etc).

Note: For Huawei / iStore the custom sensor.batteries.rated_capacity has a value of 10 kWh (that is dynamically calculated from the battery modules installed and their diagnostic reporting of their total sizes)

### Steps To Reproduce

As per above description enter your sensor containing batteries rated capacity at the two points in the visual editor

### Expected behaviour

kWh held, as shown under the battery will be 50% of what it actually is.

### Card Version

v4.39.0 (was occurring on earlier also)

### Home Assistant Version

Core: 2024.5.2/ Supervisor: 2024.05.1 / OS: 12.3

### Configuration

```shell
Example where visual editor has both boxes filled out, and same sensor in yaml under Battery and Entities sections:

type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
card_width: 50%
large_font: false
title: Huawei - Power Monitor (WLCRS v1.4.0)
title_colour: White
title_size: 18px
show_solar: true
show_grid: true
show_battery: true
decimal_places: 2

inverter:
  modern: false
  colour: grey
  autarky: energy
  auto_scale: true
  model: huawei
  three_phase: false

battery:
  invert_power: true
  colour: '#fc8d83'
  show_daily: true
  animation_speed: 5
  max_power: number.batteries_maximum_discharging_power
  show_absolute: true
  auto_scale: true
  hide_soc: true
  show_remaining_energy: true
  dynamic_colour: true
  linear_gradient: true
  animate: true
  energy: sensor.batteries_rated_capacity
  shutdown_soc: number.batteries_end_of_discharge_soc
  shutdown_soc_offgrid: number.batteries_backup_power_soc
  
solar:
  colour: '#F7BC00'
  show_daily: true
  mppts: 4
  animation_speed: 8
  max_power: 12600				  
  pv1_name: Inv1.S1
  pv2_name: Inv2.S1
  pv3_name: Inv1.S2
  pv4_name: Inv2.S2
  display_mode: 2
  auto_scale: true
  dynamic_colour: true
  pv1_max_power: 4000
  pv2_max_power: 2000
  pv3_max_power: 3300
  pv4_max_power: 3300
  efficiency: 3
  
load:
  colour: magenta
  show_daily: true
  show_daily_aux: true
  show_aux: false
  invert_aux: false
  show_absolute_aux: false
  aux_name: Generator
  aux_type: mdi:generator-portable
  aux_colour: '#5490c2'
  aux_off_colour: brown
  aux_loads: 2
  animation_speed: 4
  essential_name: Essential
  max_power: 10000				 
  additional_loads: 4
  load1_name: Servers					
  load2_name: Desktops
  load3_name: Lights
  load3_icon: mdi:lightbulb
  load4_name: All GPO
  load4_icon: mdi:power-socket-au
  auto_scale: true
  dynamic_icon: true
  dynamic_colour: true
  
grid:
  grid_name: Your-Grid-Name
  colour: '#FF2400'
  export_colour: green
  no_grid_colour: '#a40013'
  grid_off_colour: '#e7d59f'
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  invert_grid: true
  nonessential_name: Non Essential
  nonessential_icon: none
  additional_loads: 2
  load1_name: HVAC
  load2_name: EV
  load1_icon: mdi:fan
  load2_icon: mdi:car
  animation_speed: 7
  max_power: 10000
  auto_scale: true
  dynamic_icon: true
  dynamic_colour: true
  energy_cost_decimals: 3

entities:
  use_timer_248: null
  priority_load_243: null
  day_battery_charge_70: sensor.batteries_day_charge
  day_battery_discharge_71: sensor.batteries_day_discharge
  day_load_energy_84: sensor.house_consumption_energy_daily
  day_grid_import_76: sensor.hs_grid_imported_daily
  day_grid_export_77: sensor.hs_grid_exported_daily
  day_pv_energy_108: sensor.inverters_daily_yield
  day_aux_energy: sensor.sunsynk_card_aux_energy_daily
  inverter_voltage_154: sensor.inverter_phase_a_voltage												  
  load_frequency_192: sensor.power_meter_frequency
  inverter_current_164: sensor.inverter_phase_a_current
  inverter_power_175: sensor.inverters_active_power
  inverter_status_59: sensor.inverters_state
  radiator_temp_91: null
  dc_transformer_temp_90: sensor.inverters_internal_temperature
  pv1_power_186: sensor.inverter_1_pv_1_power
  pv2_power_187: sensor.inverter_1_pv_2_power
  pv3_power_188: sensor.inverter_2_pv_1_power
  pv4_power_189: sensor.inverter_2_pv_2_power
  pv_total: sensor.inverters_input_power
  environment_temp: sensor.<your_location>_temp
  remaining_solar: sensor.energy_production_today_remaining
  pv1_voltage_109: sensor.inverter_pv_1_voltage
  pv1_current_110: sensor.inverter_pv_1_current
  pv2_voltage_111: sensor.inverter_pv_2_voltage
  pv2_current_112: sensor.inverter_pv_2_current
  pv3_voltage_113: sensor.inverter_pv_1_voltage_2
  pv3_current_114: sensor.inverter_pv_1_current_2
  pv4_voltage_115: sensor.inverter_pv_2_voltage_2
  pv4_current_116: sensor.inverter_pv_2_current_2
  battery_voltage_183: sensor.batteries_bus_voltage
  battery_soc_184: sensor.batteries_state_of_capacity
  battery_power_190: sensor.batteries_charge_discharge_power
  battery_current_191: sensor.batteries_bus_current
  battery_temp_182: sensor.batteries_temperature
  battery_status: sensor.batteries_status
  essential_power: sensor.house_consumption_power_less_non_essential
  essential_load1_extra: sensor.it_hardware_energy_meter
  essential_load2_extra: null
  nonessential_power: sensor.sunsynk_card_non_essential_active_power
  grid_power_169: sensor.grid_power_169
  grid_ct_power_172: sensor.power_meter_active_power
  grid_connected_status_194: sensor.inverters_off_grid_status
  aux_power_166: sensor.sunsynk_card_aux_active_power
  aux_connected_status: binary_sensor.sunsynk_card_aux_connected_status
  energy_cost_buy: sensor.electricity_price
  energy_cost_sell: sensor.electricity_fit
  solar_sell_247: switch.null
  aux_load1: sensor.it_hardware_active_power
  aux_load2: sensor.all_desktops_active_power
  aux_load1_extra: sensor.env_network_rack_bme280_temperature
  aux_load2_extra: sensor.env_server_rack_bme280_temperature
  grid_voltage: sensor.power_meter_voltage
  battery_rated_capacity: sensor.batteries_rated_capacity
  essential_load3: sensor.lights_all_active_power
  essential_load4: sensor.gpo_all_active_power_less_known
  essential_load1: sensor.it_hardware_active_power
  essential_load2: sensor.all_desktops_active_power
  non_essential_load1: sensor.all_aircons_active_power
  non_essential_load2_extra: sensor.ev_charger_energy_daily
  non_essential_load1_extra: sensor.all_aircons_energy_daily
  non_essential_load2: sensor.pro3em_em2_active_power
dynamic_line_width: true
max_line_width: 8
min_line_width: 1
decimal_places_energy: 1
card_mod:
  style: |
    ha-card { 
      --ha-card-background: transparent;
      --ha-card-border-radius: 10px;
      --ha-card-box-shadow: 0px ;
      --ha-card-border-width: 0px;
    }
```

slipx06, 2024-05-27:
Hi The logic is as follows:

```
const battenergy = this.getEntity('battery.energy', {state: config.battery.energy?.toString() ?? ''});
let batteryEnergy = battenergy.toPower(false);
if (batteryVoltage && stateBatteryRatedCapacity.notEmpty()) {
    batteryEnergy = Utils.toNum(batteryVoltage * stateBatteryRatedCapacity.toNum(0), 0)
}
```

The `battery_rated_capacity:` sensor should provide a value in Ah. If defined together with battery voltage it calculates battery energy by multiplying the two.

slipx06, 2024-06-01:
@Roving-Ronin have you had a chance to confirm this. If you provide a sensor for battery_rated_capacity: the battery energy will be calculated as

`battery_rated_capacity x battery_voltage_183`

Roving-Ronin, 2024-06-01:
@slipx06 No sorry been busy with family / away.... will revert to you on this (and the current Power Load / Daily KWh for the whole house thing).

slipx06, 2024-06-20:
Closing this as intended logic

closed 2024-06-20 by slipx06