slipx06/sunsynk-power-flow-card #458: wrong flow color

jolly12f, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

with the latest update the flow from inverter to load is colored but when the load is powered by two sources such as photovoltaic and battery the color of the flow remains neutral

### Steps To Reproduce

the load has two power sources

### Expected behaviour

the flow could be of two colors or have the color of the more powerful one

### Card Version

Lastest

### Home Assistant Version

Lastre

jolly12f, 2024-05-27:
[image]

slipx06, 2024-05-27:
Currently the colour of the path will only change if the load or battery is being supplied 100% by one source i.e. grid or solar. Where there is a combination of two sources it reverts to the colour defined in your config

jolly12f, 2024-05-27:
Yes, I saw, could it be modified with the two colors or with the one that exceeds 50%?

closed 2024-05-29 by slipx06

jolly12f, 2024-05-29:
I can't find where to change this parameter

slipx06, 2024-05-29:
[image]

Under load and battery

```
battery:
  path_threshold: 100
load:
  path_threshold: 100
```

jolly12f, 2024-05-29:
[image]

jolly12f, 2024-05-29:
Ok i have find

jolly12f, 2024-05-29:
> ![image](https://private-user-images.githubusercontent.com/7227275/334938491-362e952e-62a9-42d4-92a1-0823b03da262.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

the non-use range should be adjusted in the grid, not just a minimum threshold because when the energy is introduced the value is always negative even if it is a value close to 0 it is not to be considered