slipx06/sunsynk-power-flow-card #460: Batteri SOH

kisjeth, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I consider the card extremely valid and with possible further implementations

### Expected behaviour

Would it be possible to also include the SOH in the battery section?

### Mode

full

### Context / Reason

For owners of storage batteries, this would be important data that they can always have under control.

slipx06, 2024-06-01:
Thanks but where would you want to display that?

[image]

kisjeth, 2024-06-01:
Yes, where you marked, or above, although I realize that each configuration can vary and so your choice may be the best. Alternatively you could opt to choose which parameter to display between the SOC shutdown or the SOH

[image]

slipx06, 2024-06-01:
That space is currently used to show battery temperature. Will see what I can do

closed 2024-06-01 by slipx06

slipx06, 2024-06-01:
Added but please note you can chose to display either battery temperature or battery SOH but not both. They are displayed in the same place on the card. If battery_temp_182: is defined it will take priority and this sensor will not be displayed.

commit 57c56924 (2024-06-01): feat: add battery SOH closes #460