slipx06/sunsynk-power-flow-card #467: Load Box color switched for Load2 and Load3

Scoperius, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Box color for Load2 is based on Load3 value and for Load3 is based on Load2 value 

### Steps To Reproduce

Last version available for today. Used is Lite card version

### Expected behaviour

Load2 box color based on Load2 value
Load3 box color based on Load3 value

### Card Version

Last

### Home Assistant Version

Last

slipx06, 2024-06-02:
@Scoperius I cant reproduce this. Everything looks good on my end. Please share your config or more info

  ```
essential_load1: sensor.sunsynk_pv1_power
  essential_load2: sensor.sunsynk_pv2_power
  essential_load3: input_number.test_number
  essential_load4: sensor.sunsynk_pv3_power
```

[image]

Scoperius, 2024-06-02:
Hi, I have only 3 loads. see the configuration:
```
load:
  show_daily: true
  additional_loads: 3
  load1_name: IVT AirX-130
  load1_icon: mdi:heat-pump-outline
  load2_name: 'Bazén '
  load2_icon: mdi:pool
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
  load3_name: Výřivka
  load3_icon: mdi:pool-thermometer
```
[image]

```
  essential_load2: sensor.pl06_plug_tu_power
  essential_load2_extra: sensor.filtrace_spotreba
  essential_load3: sensor.vyrivka_denni_spotreba
```

slipx06, 2024-06-02:
Found the problem. Thanks for reporting

closed 2024-06-02 by slipx06

Scoperius, 2024-06-04:
Now the box for Load2 is not colored at all.

Scoperius, 2024-06-04:
[image]

Scoperius, 2024-06-04:
> Have you defined an off_threshold?

Hi,

No off_threshold set.

Petr

------ Původní zpráva ------
Od "slipx06" ***@***.***>
Komu "slipx06/sunsynk-power-flow-card" 
***@***.***>
Kopie "Scoperius" ***@***.***>; "Mention" 
***@***.***>
Datum 04.06.2024 14:32:24
Předmět Re: [slipx06/sunsynk-power-flow-card] Load Box color switched 
for Load2 and Load3 (Issue #467)

Scoperius, 2024-06-04:
When > 1 box color is back

------ Původní zpráva ------
Od "Petr Lustyk" ***@***.***>
Komu "slipx06/sunsynk-power-flow-card" 
***@***.***>; 
"slipx06/sunsynk-power-flow-card" 
***@***.***>
Kopie "Scoperius" ***@***.***>; "Mention" 
***@***.***>
Datum 04.06.2024 19:03:44
Předmět Re[2]: [slipx06/sunsynk-power-flow-card] Load Box color switched 
for Load2 and Load3 (Issue #467)

slipx06, 2024-06-04:
It depends on your unit of measurement. The calculation will always try convert to W or Wh before checking if > than off_threshold. If you don't provide an expected  unit of measurement i.e. W, kW,  Wh, kWh then it will simply round that value and then do the calculation i.e. 01-0.4 will round down to 0, 0.5-0.9 will round up to 1.

purcell-lab, 2024-06-04:
Thanks for this PR. I was just about to implement a off threshold in my power sensor template for each of my devices, but I see this capability is inbuilt. 

Brilliant!

Scoperius, 2024-06-05:
It will be nice to have possibility to set off_treshold for some decimal 
points, for instance 0.1 will suit my personal need.

------ Původní zpráva ------
Od "Mark Purcell" ***@***.***>
Komu "slipx06/sunsynk-power-flow-card" 
***@***.***>
Kopie "Scoperius" ***@***.***>; "Mention" 
***@***.***>
Datum 05.06.2024 0:43:16
Předmět Re: [slipx06/sunsynk-power-flow-card] Load Box color switched 
for Load2 and Load3 (Issue #467)

commit 0f8bbc8c (2024-06-02): fix: wrong colours when displayng 3 essential loads on compact/lite card closes #467