slipx06/sunsynk-power-flow-card #468: Load3 always rounded for 2 places

Scoperius, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Load3 value rounded to 2 decimal places regardless entity is defined to be rounded for 1 decimal place and there is no possibility how to change. 

### Expected behaviour

Either to use the rounding as defined for the entity or add the possibility to change per user requirement. 

### Card Version

Last

### Home Assistant Version

Last

slipx06, 2024-06-02:
Is this essential or nonessential load? What sensor are you using? Whats the unit of measurement. Do you have auto scale enabled?

Scoperius, 2024-06-02:
Hi, I have only 3 loads. see the configuration:
```
load:
  show_daily: true
  additional_loads: 3
  load1_name: IVT AirX-130
  load1_icon: mdi:heat-pump-outline
  load2_name: 'Bazén '
  load2_icon: mdi:pool
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
  load3_name: Výřivka
  load3_icon: mdi:pool-thermometer
```
```
  essential_load3: sensor.vyrivka_denni_spotreba
```

It is Daily consuption in kWh for whirlpool

[image]

slipx06, 2024-06-02:
I think it might be something to do with your sensor. Can you test with a different sensor because the logic for the loads is all the same and when I provide a test sensor the decimal places work

[image]

[image]

Scoperius, 2024-06-02:
Decimal_places settings was missing in my code in the general configuration option. 

Thanks for help. 

You can still thing to have the possibility to override this general settings. :-)

slipx06, 2024-06-02:
The default value is 2 decimal places but you can set it to whatever you like.

closed 2024-06-02 by slipx06