slipx06/sunsynk-power-flow-card #469: Add the direction of the energy flow from the grid

matomatusov, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

If the power is taken from the grid, the direction is not displayed.
If I send power to the grid, the direction is displayed.

### Expected behaviour

Would it be possible to add the direction of the energy flow from the grid?
[image]

### Card Version

4.40.1

### Home Assistant Version

2024.5.4

slipx06, 2024-06-03:
I guess the assumption was that most people don't feed back to the grid so the normal grid icon would be sufficient.

matomatusov, 2024-06-03:
If there is no solar or battery power, the icon could show that the direction is from the grid.
The current grid icon can also remain.

closed 2024-06-03 by matomatusov

slipx06, 2024-06-05:
In the latest [release](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.42.0) you can achieve this by adding this to your config:

```
grid:
  import_icon: mdi:transmission-tower-export
```

commit ae6eab51 (2024-06-05): feat: display extra load information on compact card when additional loads = 3 and 4 closes #472
feat: customise the grid icons. See `import_icon:`, `export_icon:`, and `disconnected_icon:` under grid section. Specify mdi or sensor that returns an mdi:icon closes #475 and closes #469
fix: position of grid provider name when Daily Grid Buy is hidden