slipx06/sunsynk-power-flow-card #471: Add full support for Lifelynk

jon0881, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Almost all values are matched and pulled from the Sunsynk connect portal but there seems to be a missing value when it comes to pulling the grid and load values. I can see the values in the sunsynk connect app but the matched values in the card in HA are 0 value

[image]

### Expected behaviour

grid and load values to be populated

### Mode

Both

### Context / Reason

maybe have lifelynk as an inverter option if the data needs to be matched to a different entity?

slipx06, 2024-06-03:
Hi. The card will display the provided sensor. Make sure you have the correct sensors defined for:

```
grid_ct_power_172:
essential_power:
```

slipx06, 2024-06-05:
Also if you have a set of Lifelynk inverter codes and an image you would like displayed please share

jon0881, 2024-06-05:
Im happy to share those but im not sure where to find them. I did log a ticcket with Sunsynk as i think even in thier portal some data is not matching to the right place. For example on the input you can see power and PV1 power but not the volts or the amps
[image]