slipx06/sunsynk-power-flow-card #472: Compact Mode / Display essential_load1_extra, essential_load2_extra, essential_load3_extra, essential_load4_extra

MiniBlister, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In the compact mode the essential_load1_extra, essential_load2_extra, essential_load3_extra, essential_load4_extra is not shown

[image]

### Expected behaviour

It seems like there is space to show the values. 

### Possible Solutions

In case Load1 and Load2 would be moved up by 20 the extra field could be displayed as well.

[image]

### Mode

lite

### Context / Reason

add additional space between. 20 px should be enough and display the extra fields

MiniBlister, 2024-06-03:
Ahh, just realized, that extra3 and extra4 do not exist yet

slipx06, 2024-06-03:
At the moment extra information is only displayed when you have 2 loads. I never added them because of space but I can relook at it.

MiniBlister, 2024-06-03:
I added load3 and load4 in local copy but still need to work the position. Do you want me to commit the change in case done?

slipx06, 2024-06-03:
Sure. Happy for you to submit a pull request

MiniBlister, 2024-06-05:
submitted a pull request. Hope this works. Please let me know in case I did something wrong

PR #476 (2024-06-05): Updated Card View Compact to show essential_load3_extra and essential_load4_extra

## Description

## Related issues
closes #472 

## Motivation and Context

## How has this been tested

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] This change requires a documentation update

## Checklist

- [x] My code follows the style guidelines of this project [... cut]
commit ae6eab51 (2024-06-05): feat: display extra load information on compact card when additional loads = 3 and 4 closes #472
feat: customise the grid icons. See `import_icon:`, `export_icon:`, and `disconnected_icon:` under grid section. Specify mdi or sensor that returns an mdi:icon closes #475 and closes #469
fix: position of grid provider name when Daily Grid Buy is hidden
closed 2024-06-05 by slipx06