slipx06/sunsynk-power-flow-card #475: Generator alternative to Grid

tiaanv, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Card does not cater for generator input

### Expected behaviour

My system is completely off-grid.    Instead, I have a generator connected to my inverters.  Would be great to be able to either add generator support, or simply for now make the Grid options cater for a different icon on the grid side.

### Possible Solutions

Changeable Grid Icon.
Separate settings for Generator input.

### Mode

Both

### Context / Reason

This is probably only applicable to a very small subset of people, so not expecting too much effort/support for this.  I could fork, and dive into this myself, but then it becomes a pain to maintain, if this does gain some favour, I am happy to do some potential dev as a starting point for a PR.

slipx06, 2024-06-04:
Hi. The card does support generator input. You should use the aux input.

[image]

tiaanv, 2024-06-04:
Thx, for the feedback.  
Yes, I know.  
I really did not state my requirement well enough.

I prefer the "Lite" option for an overview, and in that mode the AUX is not shown.  Grid however is.

Interestingly, I have come across an off-grid install, where the Grid input was used for the Generator, and the AUX/GEN input was used in smartload config.

I still appreciate, that this could be a silly requirement, but all it really boils down to, is being able to change the icon of the Grid.

slipx06, 2024-06-04:
Hi. Ok I understand the request. I'll take a look.

closed 2024-06-05 by slipx06

slipx06, 2024-06-05:
@tiaanv Added to the latest release.  Please test

```
grid:
  import_icon: mdi:generator-mobile
  export_icon: mdi:generator-mobile
  disconnected_icon: mdi:generator-mobile
```

tiaanv, 2024-06-06:
I updated, it worked.  Could even set the icons from the frontend,  but then for some reason today, even though I have the yaml in there, it does not work, and the front-end options are reverted to the old one...  Not sure if this has to do with HA updating...

Either way, It did work, and perfectly addressed my requirement.  I really appreciate the change...  Wanted to post some screenshots, but now it doesn't work.  Will fiddle and report back when I can.

As a side note, but I could log another issue for that separately, the NAME is static, so when I enable Daily Import, it would say "Daily Grid Buy".  Ideally this should use the NAME as provided in the Grid name?

slipx06, 2024-06-06:
Ok thanks for the feedback. It might be a caching issue. Not sure. Try clear your browser cache or incognito mode.  Might be better to just change that to "Daily Import" and "Daily Export"

tiaanv, 2024-06-07:
Got to be a caching issue. the "Daily Import" option is perfect.

slipx06, 2024-06-08:
@tiaanv You can set custom labels in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.43.0

tiaanv, 2024-06-08:
[image]
Everything 100% I hoped for.  Thank you!!! Enjoy the coffee! ;)

slipx06, 2024-06-08:
Thanks for the coffee 😄

commit ae6eab51 (2024-06-05): feat: display extra load information on compact card when additional loads = 3 and 4 closes #472
feat: customise the grid icons. See `import_icon:`, `export_icon:`, and `disconnected_icon:` under grid section. Specify mdi or sensor that returns an mdi:icon closes #475 and closes #469
fix: position of grid provider name when Daily Grid Buy is hidden