slipx06/sunsynk-power-flow-card #477: Multiple investors

Tiaan127, 2024-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Only able to add one inverters info

### Expected behaviour

Able to add more than one inverters info in batter and inverter 

### Possible Solutions

Have them show as separate inverters or simply ad field and calculate the total for inverters and average the temperature 

### Mode

Both

### Context / Reason

One inverters data is insufficient to calculate usage

Pho3niX90, 2024-06-09:
Whilst this is possible, it's difficult to test, without having multiple inverters on hand. 

Will have a look at this as I have noticed it has come up a few times.

Roving-Ronin, 2024-06-19:
@Pho3niX90

Already raised in: https://github.com/slipx06/sunsynk-power-flow-card/issues/195 and https://github.com/slipx06/sunsynk-power-flow-card/issues/285

closed 2024-06-20 by slipx06