slipx06/sunsynk-power-flow-card #478: Off Grid Icon

bazcurtis, 2024-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

It shares the export_icon: mdi:transmission-tower-export icon

### Expected behaviour

I would like it to use mdi:transmission-tower-off

### Possible Solutions

Add an off_grid_icon option. I thought the disconnected_icon would provide this option

### Mode

Both

### Context / Reason

I think that would cover all options.

slipx06, 2024-06-08:
Hi. That icon will only display when your grid power = 0 and and the grid_connected_status_194: is reporting the correct grid disconnected state

[image]

bazcurtis, 2024-06-09:
Thanks, I will look into that. I have a binary sensor that I made already. I assume 0 watts was enough. I did not have 194 in my config.

slipx06, 2024-06-09:
The reason it's needed because you can still have your grid connected but you are just not drawing any power.

bazcurtis, 2024-06-09:
It is working fine now. Thanks for the help.

closed 2024-06-09 by bazcurtis