slipx06/sunsynk-power-flow-card #483: Add the current ' Total Power / Load' to card

Roving-Ronin, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently the card doesn't show the overall total consumption / load of power that is being consumed (at that moment).
Instead we only see the totals of Active Power coming from the grid, inverter (PV+Battery) and aux.

### Expected behaviour

Show 'Total Power / Load' that is being consumed in the house at any moment 

### Possible Solutions

Doubt it could fit, but utopian would be above the essential 1 and essential 2 boxes on the top right of the card  [showing 244W / 165W], so it could be a 2.30kW / 34.5kWh with a smaller 'Load / Daily Energy' underneath it) that would be like the 'Daily Solar / Left Today' on the left top of the card.

[image]

[image]

[image]

Alternatively the Active Load might be placed above the total energy:
[image]

Active Load / Daily Cons. are other descriptors that could be used to reduce the letter count in those examples also.?

### Mode

Both

### Context / Reason

Provide the total Power Consumption / Load being used at 'that moment' in the house, versus trying to manually add Inverter + Grid + AUX + Battery together, and by the time your mid way though that, find the readings have changed anyhow.

slipx06, 2024-06-19:
Not sure about this. That aggregated view is probably better displayed on [Power Flow Card Plus](https://community.home-assistant.io/t/power-flow-card-plus/552326) 

I assume this is over a 24h period. Total active power/energy would = essential + non_essential + Aux. Its really only non_essential for which there is no daily energy displayed on the card. The others have daily energy and active power  

[image]

Roving-Ronin, 2024-06-20:
Hi @slipx06 not energy, I was referring to power. So having the total power that is being used in the house (essential/ non essential/ aux) and from all sources, at that instant.

Roman-F-EF, 2025-02-16:
Sorry wenn ich hierzu etwas schreibe. 
Die Gesamtlast kann natürlich angezeigt werden, wenn man dafür den richtigen Sensor ausliest. Nicht jeder Wechselrichter stellt diese Werte zur Verfügung. Bei Deye zum Beispiel kann der Wert des CT, auch ext. CT gesondert ausgelesen werden, bei meinem Growatt habe ich nur die AC-Leistung des Wechselrichter und muss den CT-Wert am Übergabepunkt zum Netz gesondert erfassen. 

Ich habe selber ein verzweigtes Hausnetz mit 2 Wechselrichtern und 3 Mikrowechselrichter, die alle vor dem Netzzähler einspeisen. Ich kann also bei Deye den externen CT-Wert in die Karte als Entitität eingeben oder ich benutze einen extra Zähler fürs Haus, muss also die Werte gesondert bereitstellen. 
Es ist kein Fehler der Karte.